google/openhtf #224: Front end TestStartPrompt triggers start of test before user press Enter to start

agdeepak, 2016-04-12:
Intermittent issue:
In the front-end, PromptForTestStart trigger waits for the user to enter the DUTID but sometimes, the test starts immediately after user input and not wait for user to press ENTER to continue. This is an intermittent issue as sometimes the trigger waits for user to press ENTER to start the test and sometimes the test starts as soon as DUT ID is scanned.

Impact:
If the DUT_ID is not correct or entered wrong, then the test will trigger to start and user has no control to edit the DUT_ID or setup the DUT before test start.

jettisonjoe, 2016-04-13:
Interesting. I don't see a way for this to happen without someone clicking the ENTER button or pressing the ENTER key while the prompt form field is in focus. More investigation needed.

fahhem, 2016-04-13:
It may be possible that​ the scanner used hits enter on its own? Some can
be configured to do so

jettisonjoe, 2016-04-13:
Yeah, that sounds like it's almost certainly what's happening. @agdeepak, you probably want to double-check the configuration of all the scanners being used.

agdeepak, 2016-04-13:
Ah, I didn't know the scanners are configurable. The scanners being used were USB plug and play with no setup or configuration required. But, maybe it differs from model to model. I will have to check it when I get a chance next time around.

fahhem, 2016-04-14:
I've seen the configuration happen by scanning barcodes, so if the scanners
have been used before they may have the configuration in memory.
​

jettisonjoe, 2016-06-06:
@agdeepak, were you able to confirm whether this was a scanner issue? I'd like to close this if it turns out not to be a bug after all.

agdeepak, 2016-06-14:
I was not able to verify the scanner issue as we did not use front-end recently. We can close this bug for now and re-open if needed in future.

grybmadsci, 2016-06-14:
will do, there's a new frontend about to be launched, also, so let's close this one and reopen a new issue if there's still a problem.

closed 2016-06-14 by grybmadsci