google/openhtf #225: ENGR vs PROD output data

agdeepak, 2016-04-12:
Request for a flag or option to distinguish between ENGR and PROD runs in OpenHTF.

fahhem, 2016-04-12:
You can include this as a 'measurement' or in the metadata. I don't think OpenHTF should include this since there are so many different ways to interpret this and many teams want to do it differently

agdeepak, 2016-04-12:
This feature is basically required for end-to-end testing and data uploading setup. The data upload doesn't care about measurements pass/fail and since all PROD and ENGR data gets uploaded from the same local folder to remote, it becomes hard to distinguish between fake and real unit data records with actual DUT IDs. Metadata option might work but server will need to parse it during test record upload.

fahhem, 2016-04-12:
Wherever it goes, the server will need to parse it somehow. You can decide to make it a flag and have it go in the filename that gets created, or it can come in from a prompt and go into a measurement, but that's entirely up to you.

agdeepak, 2016-04-12:
Ok, let me explore this option offline and I will get back on this issue. I will mark it closed if it gets resolved by enabling this in measurement or a flag option.

wallacbe, 2016-04-12:
How about embedding prod vs eng into the serial number?

jettisonjoe, 2016-04-12:
I believe this came up before when we were talking about data collection and my advice then was to use an entry in the 'metadata' field in the test record. Our data team might have even already verbally agreed that they could parse it from there.

agdeepak, 2016-04-13:
That is correct, Joe. We did talk about this but I wanted to add this issue here to track it until closure and to explore any other alternate solutions (like above proposed solutions). I can followup on metadata or measurements field options and close the issue once we have a solution in place.

jettisonjoe, 2016-06-06:
@agdeepak, any progress on this issue? Have you been able to take advantage test metadata to differentiate test run types?

jettisonjoe, 2016-08-25:
As of now the recommended fix is to use test metadata. Marking closed.

closed 2016-08-25 by jettisonjoe