google/openhtf #230: Exceptions raised getting code info should not crash the framework.

grybmadsci, 2016-04-20:
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "build/bdist.linux-x86_64/egg/tbtech/tests/metrology.py", line 78, in <module>
  File "/usr/local/lib/python2.7/dist-packages/openhtf/util/measurements.py", line 383, in decorate
    phase = openhtf.PhaseInfo.WrapOrCopy(wrapped_phase)
  File "/usr/local/lib/python2.7/dist-packages/openhtf/__init__.py", line 312, in WrapOrCopy
    func = cls(func, test_record.CodeInfo.ForFunction(func))
  File "/usr/local/lib/python2.7/dist-packages/openhtf/io/test_record.py", line 98, in ForFunction
    source = inspect.getsource(func)
  File "/usr/lib/python2.7/inspect.py", line 701, in getsource
    lines, lnum = getsourcelines(object)
  File "/usr/lib/python2.7/inspect.py", line 690, in getsourcelines
    lines, lnum = findsource(object)
  File "/usr/lib/python2.7/inspect.py", line 538, in findsource
    raise IOError('could not get source code')
IOError: could not get source code

grybmadsci, 2016-04-20:
I only see this outside a virtualenv, haven't investigated the root cause yet, but at the very least this should be a warning a a fatal crash and burn.

fahhem, 2016-04-21:
Even the python stacktrace couldn't get the source code, and it looks like you might be doing something fancy with compiled python or a ZipImporter? I agree, we should wrap the getsource functions with a try-except. I can take this one if you haven't started it

grybmadsci, 2016-04-21:
I don't think we were doing anything fancy, just running on an ubuntu install, but outside a virtualenv.

jettisonjoe, 2016-04-21:
Outside a virtualenv, meaning using the default system python?

grybmadsci, 2016-04-21:
Yep, default system python on Ubuntu.

closed 2016-04-23 by grybmadsci