google/openhtf #240: add plug takes too long to pull file (size larger than 3Mega byte)

fanchang0722, 2016-04-30:
When using adb plug to pull large data file (filesize >3M), it takes 3x time longer than using "adb pull filename" directly. Could you help to take a look? Thanks.

fahhem, 2016-05-02:
Can you try using the code in the `python-adb` project? We addressed some problems in there that may be related

jettisonjoe, 2016-06-06:
So after following up with @fanchang0722  in person, it sounds like `python-adb` is doable. Although it's pulls on large files still aren't as fast as normal command-line `adb`, they are a lot faster than ours. I think we should consider re-writing our `AdbPlug` plug to use `python-adb` and ripping out our own implementation.

fahhem, 2016-06-06:
I'm working on adding parallel connections to `python-adb` (in my copious free time) in a way that won't add overhead for singular connections, which should keep the speed the same.

@fanchang0722 can you provide an example file and benchmark so I can add it to `python-adb`? That way I can ensure performance at least doesn't get any worse in the future.

grybmadsci, 2016-06-06:
+1, switching to python-adb SGTM.  that also makes our adb plug a thin wrapper around that library.

fanchang0722, 2016-06-07:
Sure. Let do it this week. Thanks 

Fan