google/openhtf #243: Add finally block for phases

wallacbe, 2016-05-06:
There is a phase I would like to execute at the end of the tester to collect logs, especially if an error is raised.  Currently test execution ceases if an error is raised.  Can a finally block be added for such phases?

grybmadsci, 2016-05-06:
Hm, you *could* use an output callback for this sort of thing, but it's generally bad practice to modify the output in the output callback, so it's probably not the best solution.  Does your finally block need to use plugs?

fahhem, 2016-05-06:
From what he's mentioned, and also for my use case, I would say yes. It should
run right before the plugs are torn down. More concrete use cases include a
background thread listening to serial logs, load generators, etc. OpenHTF
doesn't allow altering measurements from a background thread (the test data's
measurements change as the phases start/end, so you can't access them anymore),
so the next best thing is to collect them all at the very end.
An alternative to some of the cases is to allow measurements to be captured in
the background, but not for all since you may want to record “last current draw”
as a data point for failures for instance. The latter has come up in our
load-generating tests, as we'd like to know if it failed open or closed, but
still end the test immediately so the tear down methods shut off power, etc to
prevent cascading failures.

grybmadsci, 2016-05-06:
Ah, yeah this came up when `@monitors` was added, because it only works at a per-phase level, there is currently no test-wide `@monitors` equivalent.

I'll put some thought into it, but we do generally want to minimize test-wide concepts as much as possible, including keeping test flow control as simple as we can.  That said, a test teardown phase sounds like it would get used enough to warrant adding, so we'll have to come up with something, it's just unclear exactly what that looks like.

fahhem, 2016-05-06:
My thoughts were to create an object that you'd wrap the inner phases in:
phases = [ PhaseOne, TryBlock([ TryPhaseOne, TryPhaseTwo, ], Except([ ExceptPhaseOne, ]), Else([ ElsePhaseOne ]), Finally([ FinallyPhaseOne, ]) ]
This would mirror python syntax/indentation without stepping on reserved names
(caps-sensitivity FTW). Except/Else could be removed to make only a TryBlock and
Finally.
Implementing it was going to be a “replace PhaseExecutor entirely” thing, which
was also part of my separate execution from initialization push, so I could
create a Test and replace the bits I wanted without affecting interfaces/the UI.

grybmadsci, 2016-05-06:
Hm, I'd rather keep the syntax much simpler than that, I was thinking of having the ability to specify a `tear_down=TearDownPhase` kwarg to the `Test` constructor, and it's a phase like any other, it just is guaranteed to always be called at the end of the test (under normal execution, getting killed obviously can prevent it).

We've steered away from flow control between phases very intentionally (`run_if` being the most complex such mechanism we support), and full try/except/finally/else support is even more complex than simple branching (which was determined to be too much).

Thoughts, @jettisonjoe ?

wallacbe, 2016-05-06:
For my use cases I would need access to plugs.

grybmadsci, 2016-05-09:
Oh, it wouldn't be a kwarg to `Test`, it would go into `.Configure()` as a test-wide configuration option.  Chatted with Joe about this, and we both want to avoid the addition of flow control at the phase level.  I'd like to propose a paradigm like this:
```
def MyPhase(test):
  # Do some stuff
  raise MyError()

@plug(foo=foo_plug.FooPlug)
@measures('bar')
def TearDown(test, foo)
  test.measurements.bar = foo.CollectData()

if __name__ == '__main__':
  test = Test(MyPhase)
  # TearDown phase will always be run at the end of the test, even if an error occurs.
  test.Configure(tear_down=TearDown)
  test.Execute()
```
I'm in the process of putting together some minor changes to the `PhaseData` object passed into phases as the first arg - I can also add the ability to query some information about test state, so you could query for whether or not an exception was raised in the `TearDown` phase.

Thoughts?  I'm open to alternatives, but they have to be syntactically straightforward and can't inject additional flow control mechanics for phases.

jettisonjoe, 2016-05-10:
I agree that full phase flow control is currently beyond the scope of what OpenHTF purports to do, by design. I'm open to bring able to specify a tear-down function in `Configure()` and I like the idea of giving it the same arguments that phases normally get, so using it will feel familiar.

On the issue of full flow control, one crazy idea to enable that (without adding essentially a flow control language into OpenHTF itself) would be to abstract out some of the lower-level functions of OpenHTF to create alternatives to the monolithic `Execute()`, like `ExecutePhase(...)`, and let test authors do flow control natively in Python.

wallacbe, 2016-05-21:
Documenting workaround that was provided in email

```
class TearDownExecutor(TestExecutor):

  def __init__(self, test_data, plug_manager, teardown_phase):
    super(TearDownExecutor, self).__init__(test_data, plug_manager)
    self._teardown_phase = openhtf.PhaseInfo.WrapOrCopy(teardown_phase)

  def _ExecuteTestPhases(self, executor):
    """Executes one test's phases from start to finish."""
    self._test_state.SetStateRunning()
    for phase_outcome in executor.ExecutePhases():
      if self._test_state.SetStateFromPhaseOutcome(phase_outcome):
        break
    else:
      self._test_state.SetStateFinished()
    # Run tear down phase.
    self._test_state.pending_phases.insert(0, self._teardown_phase)
    executor._ExecuteOnePhase(self._teardown_phase)          
```      

In tester:
`  test._executor = exe.TearDownExecutor(test.data, plugs.PlugManager(), TearDownPhase)
`

fahhem, 2016-05-23:
On second thought (and implementation), the solution I'm going to be using to to tag a phase as needing another phase for cleanup:

```py
@plugs.plug(Serial)
def CloseSerial(test, serial):
  ...

@Finally(CloseSerial)
@plugs.plug(Serial)
def OpenSerial(test, serial):
  ...
```

That means I just put `OpenSerial` in my phases list and `CloseSerial` will automatically get included at the end (via tear down phase, etc).

wallacbe, 2016-05-23:
How would multiple and duplicate tags behave with this implementation idea?  Would it also have to reach the phase in the executor to run?  Thanks

@plugs.plug(ps)
def PowerSupplyTearDown(test, ps):

@plugs.plug(serial)
def SerialTearDown(test, serial):

@Finally(PowerSupplyTeardown)
@plugs.plug(ps)
def ReadCurrent1(test, ps):

@Finally(PowerSupplyTeardown)
@plugs.plug(ps)
def ReadCurrent2(test, ps):

@Finally(SerialTearDown)
@plugs.plug(serial)
def Stuff(test, serial)

fahhem, 2016-05-26:
Each phase with a finally entered will push the finally phase onto the stack. If a finally phase is already in the stack, then it's not pushed, but the phase is deep-compared (args may have changed).

grybmadsci, 2016-05-26:
Hm, what would go into such a `PowerSupplyTeardown` or `CloseSerial` that wouldn't go into the `TearDown` of the respective plug?

fahhem, 2016-05-26:
At the moment, it's stuff like "take logged serial and put it into a test attachment". If we reorder the operations in the test to have plugs' `TearDown` get called before the output callbacks, then I could get rid of some of that hackery.

grybmadsci, 2016-05-26:
`TearDown` does get called before output callbacks, expressly for this reason (so things that happen during `TearDown` can be logged and included in the output).

fahhem, 2016-05-26:
Ah, I was even involved in that change! Time to rewrite some tests!

wallacbe, 2016-06-05:
https://github.com/google/openhtf/pull/262

Any objections to closing issue or do we need further discussion?

closed 2016-06-05 by grybmadsci