google/openhtf #244: Clean up phase interfaces

grybmadsci, 2016-05-09:
This includes the phase data object passed into phases, and the return value from phases, (`PhaseResult.FAIL` should really be `PhaseResult.STOP` to match `PhaseResult.CONTINUE` and `PhaseResult.REPEAT`).

I'd like to know if anyone is actually using `PhaseResult` values to return from Phases, @wallacbe @fahhem @jettisonjoe ?

fahhem, 2016-05-10:
I have a REPEAT and a FAIL use already, but I'm able to update them as necessary
when I import the openhtf changes, so I'm fine with a STOP. However, would STOP
set the outcome to FAIL or ABORTED or a new STOPPED?

wallacbe, 2016-05-19:
I'm not using the PhaseResult values

jettisonjoe, 2016-05-24:
I don't think anyone in Aura is using them.

glados-verma, 2022-10-13:
Some work has been done to `PhaseResult`; it's a bit cleaner now. I know we use it in my group.

closed 2022-10-13 by glados-verma