google/openhtf #249: hello_world hangs - AttributeError: 'NoneType' object has no attribute 'GetFinishedRecord'

wallacbe, 2016-05-16:
Cloned repo from head, hello world hangs, CTRL+C shows:

python hello_world.py 
^C2016-05-16 16:25:54,533 - ERROR - Received SIGINT.
2016-05-16 16:25:54,534 - ERROR - Stopping Test due to SIGINT: None
Traceback (most recent call last):
  File "hello_world.py", line 87, in <module>
    test.Execute(test_start=triggers.PromptForTestStart())
  File "/home/source/openhtf/openhtf/__init__.py", line 181, in Execute
    record = self._executor.GetState().GetFinishedRecord()
AttributeError: 'NoneType' object has no attribute 'GetFinishedRecord'

jettisonjoe, 2016-05-16:
The error message here is a red herring that occurs if you ctrl+c before test state is created. The bigger concern is why hello_world.py is hanging for you. I just cloned from head and was able to run hello_world.py without the hang.

Can you tell me your exact steps to repro?

wallacbe, 2016-05-17:
I deleted my virtualenv and repo and followed the steps in CONTRIBUTING.md.  Found that the issue is that I do not see the prompt "Provide a DUT ID in order to start the test." which created confusion.

(venv)laptop:~/source/openhtf/examples$ python hello_world.py 
test
Hello World!
Finishing test execution with outcome PASS.
(venv)laptop:~/source/openhtf/examples$ python hello_world.py 
test
Hello World!
Finishing test execution with outcome PASS.

wallacbe, 2016-05-17:
Able to reproduce.  Through some trial and error I found I can get it to reliably print out by removing  termios.tcflush or adding a delay before termios.tcflush in ConsolePrompt.

 print self._message
 time.sleep(.01)
 # Before reading, clear any lingering buffered terminal input.
 termios.tcflush(sys.stdin, termios.TCIOFLUSH)

I also tried doing sys.stdout.flush() instead of the delay and that didn't appear to help.

jettisonjoe, 2016-05-19:
Thanks, I'll check it out.

wallacbe, 2016-05-21:
Any objections to merging any of the following?

1. comment out:
# termios.tcflush(sys.stdin, termios.TCIOFLUSH)

2. add delay?

3. Switch to sys.stdin.flush()?

grybmadsci, 2016-05-21:
I'd rather understand what's going on here a little better, and potentially switch to using readline (i believe that's what we had talked about?) as a more cross-platform solution.  Specifically, https://pypi.python.org/pypi/pyreadline is a pure-python implementation that is suggested as a Windows-friendly alternative at https://pypi.python.org/pypi/gnureadline

grybmadsci, 2016-05-21:
that works, looking into it pyreadline is overkill, mostly for history and autocompletion.  that said, i'd still like a cross platform solution that doesn't rely on termios eventually, but thanks for the fix for now.

PR #261 (2016-05-21): Use TCIFLUSH to only flush recieved but not read
According to http://linux.die.net/man/3/tcflush, tcflush cares about the 'object referred to by fd', which is shared between stdin, stdout, and stderr. Therefore, when we do a tcflush on IO right after a print, we will drop the data from the print that hasn't been received by the terminal (could be just 1 byte, the newline, and the terminal won't print anything).

By switching to TCIFLUSH, we only drop data that has been sent to us by the terminal but not read by us, such as any stray keyboard input. This shouldn't drop any data going the other  [... cut]
closed 2016-05-21 by grybmadsci