google/openhtf #263: Add measurement helper function

wallacbe, 2016-05-22:
Add helper function to help reduce boilerplate when defining measurements.  Supporting lists and single parameters would give most flexibility.

def Meas(parameters, units=None, limits=None)

PARAMS = [param1, param2, param3]
UNITS = ['volts', 'amps', None]
LIMITS = [(1,2), (None, 1), None]

Meas(PARAMS, UNITS, LIMITS)
Meas(param1, 'volts', (1,2))

grybmadsci, 2016-05-22:
Hm, I'm not sure exactly what you're asking for here.  You can already use these syntaxes:
```
@measures('param1')
def MyPhase(test):
  pass

@measures('param2', units='volts')
def MyOtherPhase(test):
  pass
```
And a handful of other shortened versions, demonstrated here:
https://github.com/google/openhtf/blob/master/examples/measurements.py

If you want to dynamically generate them from lists, you can do something like:
```
NAMES = ['one', 'two']
UNITS = ['volts', 'amps']

@measures(Measurement(name, units=units) for name, units in zip(NAMES, UNITS))
def MyPhase(test):
  pass
```

wallacbe, 2016-05-22:
ah, yeah that works thanks.  closing issue.

@measures('inline_kwargs', docstring='This measurement is declared inline!',
          units=UOM['HERTZ'], validators=[validators.InRange(0, 10)])

closed 2016-05-22 by wallacbe