google/openhtf #269: RuntimeError is raised when looping Execute

wallacbe, 2016-05-24:
I put Execute in a loop:
while True:
    test.Execute(test_start=triggers.PromptForTestStart())

Raises RuntimeError

python hello_world.py
Provide a DUT ID in order to start the test.
test
Hello World!
Finishing test execution with outcome PASS.
Traceback (most recent call last):
  File "hello_world.py", line 88, in <module>
    test.Execute(test_start=triggers.PromptForTestStart())
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/__init__.py", line 176, in Execute
    self._executor.Start()
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/__init__.py", line 89, in Start
    self.start()
  File "/usr/lib/python2.7/threading.py", line 739, in start
    raise RuntimeError("threads can only be started once")
RuntimeError: threads can only be started once

fahhem, 2016-05-24:
Dang, `exe.TestExecutor` is a thread that can't be started more than once.

wallacbe, 2016-05-24:
I have an old wheel I built 4/11 that allows me to loop.   Any way we can revert back to prior method?

grybmadsci, 2016-05-24:
Fixed, thanks farz.

PR #270 (2016-05-24): Move test executor creation back into Execute
Since it's a Thread (specifically, `KillableThread`), it cannot be started/stopped in a loop. Will need to come up with another solution for this.

Fixes #269
closed 2016-05-24 by grybmadsci