google/openhtf #271: TypeError: 'dict' object is not callable if http_port set

wallacbe, 2016-05-24:
Synced from head.  See the following if http_port is set.  I don't see if if it's set to None.

Provide a DUT ID in order to start the test.
test
Hello World!
Finishing test execution with outcome PASS.
----------------------------------------
Exception happened during processing of request from ('192.168.1.37', 32946)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 295, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 649, in __init__
    self.handle()
  File "/usr/lib/python2.7/BaseHTTPServer.py", line 340, in handle
    self.handle_one_request()
  File "/usr/lib/python2.7/BaseHTTPServer.py", line 328, in handle_one_request
    method()
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/http_api.py", line 114, in do_GET
    result = {'test': data.ConvertToBaseTypes(self.executor.GetState()),
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/data.py", line 127, in ConvertToBaseTypes
    obj = {k: ConvertToBaseTypes(v, ignore_keys) for k, v in obj.iteritems()
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/data.py", line 128, in <dictcomp>
    if k not in ignore_keys}
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/data.py", line 127, in ConvertToBaseTypes
    obj = {k: ConvertToBaseTypes(v, ignore_keys) for k, v in obj.iteritems()
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/data.py", line 128, in <dictcomp>
    if k not in ignore_keys}
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/data.py", line 116, in ConvertToBaseTypes
    obj = obj._asdict()
TypeError: 'dict' object is not callable
----------------------------------------

grybmadsci, 2016-05-24:
I blame farz :P Fixed, my bad, I should have noticed that in code review...

fahhem, 2016-05-24:
Aw dang, no idea how it got there, but yeah, my bad!

> Synced from head. See the following if http_port is set. I don't see if if
> it's set to None.
>
> Provide a DUT ID in order to start the test.
> test
> Hello World!
> Finishing test execution with outcome [...]
"/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/http_api.py",
> line 114, in do_GET
> result = {'test': data.ConvertToBaseTypes(self.executor.GetState()),
> File
"/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/data.py",
> line 127, in ConvertToBaseTypes
> obj = {k: ConvertToBaseTypes(v, ignore_keys) for k, v in obj.iteritems()
> File
"/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/data.py",
> line 128, in
> if k not in ignore_keys}
> File
"/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/data.py",
> line 127, in ConvertToBaseTypes
> obj = {k: ConvertToBaseTypes(v, ignore_keys) for k, v in obj.iteritems()
> File
"/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/data.py",
> line 128, in
> if k not in ignore_keys}
> File
"/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/data.py",

closed 2016-05-24 by wallacbe