google/openhtf #277: frontend broken because of logs.setup_logger()

dmho418, 2016-05-25:
Not sure if this has been addressed on another branch, but in master there is a call to  logs.setup_logger() in the frontend which is defined as SetupLogger() in logs
[frontend](https://github.com/google/openhtf/blob/master/openhtf/io/frontend/__main__.py#L239)
[logs.py](https://github.com/google/openhtf/blob/master/openhtf/util/logs.py#L166)

jettisonjoe, 2016-05-25:
Good eye, thanks, @dmho418. We used to have a convention of class and method names being PascalCase and function names using underscores. Looking at the [Google style guide](https://google.github.io/styleguide/pyguide.html#Naming), it looks like we actually made a mistake in having method names be PascalCase. I'll change the frontend code to use the current `SetupLogger()`, but when we do a lint-focused pass over the codebase (and tighten up our linting rules) we should strongly consider standardizing to the Google style guide. @grybmadsci

dmho418, 2016-05-26:
Thanks for checking!
By the way, I just saw you have a big pull request with a new frontend. Looking forward to that!

jettisonjoe, 2016-06-27:
Fixed via #246.

closed 2016-06-27 by jettisonjoe