google/openhtf #281: Make timeout error more specific and add ability to set a measurement

wallacbe, 2016-05-26:
When a phase hits a timeout it raises TimeoutError which isn't very descriptive.  I'd like to be able to Override this error with something more descriptive and also be able to set a measurement station to "FAIL" if this occurs.

wallacbe, 2016-05-27:
We could get around the measurement issue by setting the test.measurement.status = False at the begining of the phase.  However we need to remove the warning message that is generated by setting it again to True.

grybmadsci, 2016-06-01:
I was talking to @jettisonjoe about this a bit today.  I think what we really need is an audit of test pass/fail mechanisms and what tunable knobs we want around them, timeouts included.  Let's bring this up in the next OpenHTF meeting.