google/openhtf #290: plug object created/torn down each time when looping

wallacbe, 2016-06-02:
With the latest looping approach:
```
 while True:
    test.Execute(test_start=triggers.PromptForTestStart())
```
plug objects are created and torn down with every test run.  I'd like to confirm that this is the approach we want to take. With the prior looping mechanism the constructor was run only once.  The impact here could be additional test time if the constructor is run each time.

grybmadsci, 2016-06-02:
That is the intention, yes.  We can potentially make a configuration option to not tear down plugs until some later time (either `__del__` or an explicit method on the `Test` object) if it's really that big an issue.

jettisonjoe, 2016-06-03:
Agreed, tearing down after each run is desired because it makes it easier and more likely to have consistent plug state between test runs.

fahhem, 2016-06-03:
I've wanted a way to replace plugs in early execution, and I think this use-case falls into that.

Let's say I have a test with phases that say they want to control the power of a board. Currently, they must specify a plug that does this, and if I want to reuse those phases with another tester that, say, have N of these boards, I can't. I would like that other tester to be able to say "these phases, but replace PowerBoard with PowerFixture().ForBoard(i) for i in range(5)".

This would be relevant in the looping scenario, because I could then say "these phases, but replace PowerBoard with this instance of PowerBoard I already have laying around" and create the plugs before the first loop.

At the moment, if I do this in a way similar to the unit-tests madsci is working on, I lose the ability for the framework to catch these errors and expose them either in uploads or at least to the UI. If there was a way to replace plugs with a lamda that returns a plug instance, that would allow me to do this:

```py
expensive_plug = ExpensiveStartPlug()
openhtf.Test(
  PowerCycleBoard.ReplacePlug(PowerSupply, lambda: DualChannelPowerSupply().ForChannel(1)),
  PhaseWithExpensivePlug.ReplacePlug(ExpensiveStartPlug, lambda: expensive_plug),
)
```

Does that make sense?

jettisonjoe, 2016-06-03:
Interesting. I understand the use-case a little better now. Willing to entertain this since it seems like you guys both need it (or something like it). @grybmadsci?

grybmadsci, 2016-06-04:
Hm, I think i'm not fully understanding the use case.  If you have a `PowerBoard` plug, but you can either communicate directly with it (your first case) or grab it from some other place (your second case), aren't you still fundamentally using the same plug just changing the underlying transport used?

I'm concerned that you're now adding a level of indirection in the "figuring out what is going on" process - if you know phase X is acting weird and you look at phase X, you could easily be misled by seeing the wrong plug type there, and spend a lot of time going down misleading paths of debugging.

I would rather consider how we might leverage existing constructs, potentially to establish recommended paradigms, for these problems.

For example, you could either make the plug more configurable about how it talks to the underlying hardware using a `conf` key, or remove the plug entirely from the phase declaration and make both users decorate with the specific plug they want to use at `Test` creation time:
```
def PowerCycleBoard(test, power_supply):
  pass

def OneTest():
  openhtf.Test(
    plugs.plug(power_supply=psu.PowerSupply)(PowerCycleBoard),
    PhaseWithExpensivePlug,
  ).Execute()

def AnotherTest():
  # If you want plugs shared between tests, there's no choice but to manage its lifetime yourself.
  DualChanPsu = psu.DualChannel()
  openhtf.Test(
      *(plugs.plug(power_supply=DualChannelPsu.ForChannel(channel))(PowerCycleBoard)
        for channel in (1, 2))
  ).Execute()
```
Here, `ForChannel()` has to be designed to return the type to instantiate rather than the instance itself, which can get tricky, but the lambda mechanism won't work anyway because plug types must derive from `BasePlug`, and lambda's don't.

For an expensive plug, you could build the state stashing into the type itself, then you end up doing something like:
```
class ExpensivePlug(plugs.BasePlug):
  _instance = None

  def __new__(cls, *args):
    if cls._instance is None:
      cls._instance = super(ExpensivePlug, cls).__new__(cls, *args)
    else:
       # expensive initialization goes here
       cls._instance.foo = 'bar'  # instead of self.foo = 'bar'
    return cls.instance

  def ReallyTearDown(self):
    # Tear down that won't be called automatically by OpenHTF.
...

@plug(expensive_plug=ExpensivePlug)
def MyPhase(test, expensive_plug)
  # ...

def MyTest():
  expensive_plug = ExpensivePlug()
  try:
    openhtf.Test(...).Execute()
  finally:
    expensive_plug.ReallyTearDown()
```

jettisonjoe, 2016-10-10:
I think plug placeholders resolves the last of this discussion, so closing.

closed 2016-10-10 by jettisonjoe