google/openhtf #291: SigInt isn't stopping tester configured for looping

wallacbe, 2016-06-02:
Add loop to all_the_things:
```
 while True:
    test.Execute(test_start=triggers.PromptForTestStart())
```

python all_the_things.py 
Provide a DUT ID in order to start the test.
test
Hello World!
What's the widget type?
^C2016-06-01 19:50:24,605 - ERROR - Received SIGINT.
2016-06-01 19:50:24,605 - ERROR - Stopping Test due to SIGINT
2016-06-01 19:50:24,606 - ERROR - Test state: <TestState: test@wallaceb-glaptop Running Phase: hello_world>
Tearing down <ExamplePlug: 0>
^C2016-06-01 19:50:42,645 - ERROR - Received SIGINT.
^C

wallacbe, 2016-06-02:
I added sys.exit() at the end of Stop() in openhtf/__init__.py and I see that SIGINT stops the tester and plugs tear down methods are called.

python all_the_things.py 
Provide a DUT ID in order to start the test.
test
Hello World!
What's the widget type?
^C2016-06-01 21:35:43,877 - ERROR - Received SIGINT.
2016-06-01 21:35:43,877 - ERROR - Stopping Test due to SIGINT
2016-06-01 21:35:43,878 - ERROR - Test state: <TestState: Running Phase: hello_world>
Tearing down <ExamplePlug: 0>

fahhem, 2016-06-02:
Ah, right, it's not http_api, it's the fact that you're doing `while True`. When `Test.Stop()` gets called (by hitting Ctrl-C, we set `self._stopped = True` then if you try to do `test.Execute(...)` we check `_stopped` and refuse to run the test. So you sit in that while loop forever.

I just checked, and if we raise `KeyboardInterrupt` in `Test.Stop()` then it works. I think we should add that so I'm sending a PR for that now.

PR #295 (2016-06-05): Raise KeyboardInterrupt when catching SIGINT
Fixes #291
commit d4837eef (2016-06-05): Raise KeyboardInterrupt when catching SIGINT

Fixes #291
closed 2016-06-05 by grybmadsci