google/openhtf #292: tear down function doesn't run if SIGINT recieved

wallacbe, 2016-06-02:
running all_the_things without looping.  Plug tear down runs but tear down function doesn't run.

python all_the_things.py 
Provide a DUT ID in order to start the test.
test
Hello World!
What's the widget type?
^C2016-06-01 21:03:15,698 - ERROR - Received SIGINT.
2016-06-01 21:03:15,698 - ERROR - Stopping Test due to SIGINT
2016-06-01 21:03:15,698 - ERROR - Test state: <TestState: test@wallaceb-glaptop Running Phase: hello_world>
Tearing down <ExamplePlug: 0>

fahhem, 2016-06-02:
I don't see this, are there any changes in your branch from master?

wallacbe, 2016-06-02:
I synced from head again this morning and added https://github.com/google/openhtf/pull/295.  I don't see the tear down from the phase (I only see the plug tear down)

Provide a DUT ID in order to start the test.
test
Hello World!
What's the widget type?
^C2016-06-02 11:50:43,015 - ERROR - Received SIGINT.
2016-06-02 11:50:43,015 - ERROR - Stopping Test due to SIGINT
2016-06-02 11:50:43,015 - ERROR - Test state: <TestState: test@wallaceb-glaptop Running Phase: hello_world>
Tearing down <ExamplePlug: 0>
Traceback (most recent call last):
  File "all_the_things.py", line 116, in <module>
    test.Execute(test_start=triggers.PromptForTestStart())
  File "/home/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/__init__.py", line 181, in Execute
    self._executor.Wait()
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/__init__.py", line 105, in Wait
    self.join(sys.float_info.max)  # Timeout needed for SIGINT handling.
  File "/usr/lib/python2.7/threading.py", line 960, in join
    self.__block.wait(delay)
  File "/usr/lib/python2.7/threading.py", line 358, in wait
    _sleep(delay)
  File "/home/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/__init__.py", line 143, in StopFromSigInt
    raise KeyboardInterrupt
KeyboardInterrupt

wallacbe, 2016-06-08:
Can we add this to PyPi tracker?  Thanks

maddychan, 2016-07-19:
PR #372 Fixed teardown bug should resolve this

PR #413 (2016-08-23): Fix *all* outstanding bugs for PyPI.
Fixes #292 
Fixed #198
closed 2016-08-23 by jettisonjoe