google/openhtf #294: BaseException type isn't caught and tester exits

wallacbe, 2016-06-02:
I raise the following exception and it isn't caught by the framework and the tester exits 

class TimeoutError(BaseException):
  pass

python all_the_things.py 
Provide a DUT ID in order to start the test.
test
Hello World!
Exception in thread hello_world:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/threads.py", line 59, in run
    self._ThreadProc()
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/phase_executor.py", line 114, in _ThreadProc
    phase_return = self._phase(self._phase_data)
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/__init__.py", line 347, in __call__
    return self.func(phase_data, **kwargs)
  File "all_the_things.py", line 56, in hello_world
    raise TimeoutError('base exception')
TimeoutError: base exception

2016-06-01 22:56:28,462 - ERROR - Failed to execute test.
Traceback (most recent call last):
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/__init__.py", line 158, in _ThreadProc
    self._ExecuteTestPhases(executor)
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/__init__.py", line 211, in _ExecuteTestPhases
    for phase_outcome in executor.ExecutePhases():
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/phase_executor.py", line 170, in ExecutePhases
    result = self._ExecuteOnePhase(self.test_state.pending_phases[0])
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/phase_executor.py", line 195, in _ExecuteOnePhase
    outcome_wrapper.SetOutcome(phase_thread.JoinOrDie())
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/phase_executor.py", line 146, in JoinOrDie
    return PhaseOutcome(threads.ThreadTerminationError())
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/phase_executor.py", line 82, in __init__
    raise InvalidPhaseResultError('Invalid phase result', phase_result)
InvalidPhaseResultError: ('Invalid phase result', ThreadTerminationError())
Tearing down <ExamplePlug: 0>
ERROR:root:Thread raised an exception: TestExecutorThread
Traceback (most recent call last):
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/threads.py", line 59, in run
    self._ThreadProc()
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/__init__.py", line 158, in _ThreadProc
    self._ExecuteTestPhases(executor)
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/__init__.py", line 211, in _ExecuteTestPhases
    for phase_outcome in executor.ExecutePhases():
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/phase_executor.py", line 170, in ExecutePhases
    result = self._ExecuteOnePhase(self.test_state.pending_phases[0])
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/phase_executor.py", line 195, in _ExecuteOnePhase
    outcome_wrapper.SetOutcome(phase_thread.JoinOrDie())
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/phase_executor.py", line 146, in JoinOrDie
    return PhaseOutcome(threads.ThreadTerminationError())
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/phase_executor.py", line 82, in __init__
    raise InvalidPhaseResultError('Invalid phase result', phase_result)
InvalidPhaseResultError: ('Invalid phase result', ThreadTerminationError())
Exception in thread TestExecutorThread:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/util/threads.py", line 59, in run
    self._ThreadProc()
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/__init__.py", line 158, in _ThreadProc
    self._ExecuteTestPhases(executor)
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/__init__.py", line 211, in _ExecuteTestPhases
    for phase_outcome in executor.ExecutePhases():
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/phase_executor.py", line 170, in ExecutePhases
    result = self._ExecuteOnePhase(self.test_state.pending_phases[0])
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/phase_executor.py", line 195, in _ExecuteOnePhase
    outcome_wrapper.SetOutcome(phase_thread.JoinOrDie())
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/phase_executor.py", line 146, in JoinOrDie
    return PhaseOutcome(threads.ThreadTerminationError())
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/phase_executor.py", line 82, in __init__
    raise InvalidPhaseResultError('Invalid phase result', phase_result)
InvalidPhaseResultError: ('Invalid phase result', ThreadTerminationError())

Traceback (most recent call last):
  File "all_the_things.py", line 121, in <module>
    test.Execute(test_start=triggers.PromptForTestStart())
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/__init__.py", line 193, in Execute
    record = self._executor.GetState().GetFinishedRecord()
  File "/home/ /source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/test_state.py", line 160, in GetFinishedRecord
    'Internal framework error, test outcome unset!')
openhtf.exe.test_state.FrameworkError: Internal framework error, test outcome unset!

fahhem, 2016-06-02:
The python exception hierarchy works like so:
```
BaseException
 +-- SystemExit
 +-- KeyboardInterrupt
 +-- GeneratorExit
 +-- Exception
```
Continued at https://docs.python.org/2/library/exceptions.html#exception-hierarchy

So you're not supposed to catch `BaseException` (nor raise it or a direct subclass), but `Exception`. By doing `except Exception` like we do in most places, we're catching and handling errors like `AssertionError` or `EOFError` but allowing `SystemExit` (raised when another thread calls `sys.exit()`) to work normally.

If you're creating an exception, you should subclass `Exception` (or another subclass) directly rather than `BaseException`, `KeyboardInterrupt`, etc.

wallacbe, 2016-06-02:
Acknowledged, I'll request changes in the culprit library that openhtf is calling

jettisonjoe, 2016-06-03:
If we're requesting a fix in whichever library is raising this, is it still an OpenHTF bug?

fahhem, 2016-06-03:
No, but neither @wallacbe nor I can close bugs

closed 2016-06-03 by jettisonjoe

wallacbe, 2016-06-04:
I think originator can close bugs (which was me)