google/openhtf #304: setup.py isn't locating protoc on OSX

wallacbe, 2016-06-06:
setup.py isn't locating protoc on OSX

Homebrew default on OSX /usr/local

fahhem, 2016-06-07:
what does `pkg-config --variable prefix protobuf` output on OSX?

areusch, 2016-06-07:
pkg-config is not installed by default on OS X; the subprocess call in setup.py throws `OSError`, not `CalledProcessError`. homebrew installs protoc into `/usr/local/bin` by default, so it would be sane to look there if you are guessing. 

if you run `brew install pkg-config` and have previously run `brew install protobuf` then the pkg-config call succeeds and the wheel builds properly.

areusch, 2016-06-07:
Here is the output from my box when run w/o pkg-config:
```

  Running setup.py bdist_wheel for openhtf
  Complete output from command /usr/bin/python -c "import setuptools;__file__='/var/folders/kw/8z1zw32n5v9b1pdt22_qp1wm002ygm/T/pip-_Sbdu9-build/setup.py';exec(compile(open(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" bdist_wheel -d /var/folders/kw/8z1zw32n5v9b1pdt22_qp1wm002ygm/T/tmpr7ueaYpip-wheel-:
  running bdist_wheel
  running build
  running build_proto
  error: [Errno 2] No such file or directory
```

fahhem, 2016-06-07:
Okay, #305 is no longer tentative, it should definitely fix it. It will catch OSError for when pkg-config doesn't exist and default to /usr/local

PR #305 (2016-06-07): Default to /usr/local for OS X in setup.py
Fixes #304
commit 8aaece37 (2016-06-07): Default to /usr/local for OS X in setup.py

Fixes #304
closed 2016-06-07 by jettisonjoe