google/openhtf #308: Add instructions for installing protoc

wallacbe, 2016-06-07:
Add instructions for installing protoc

wallacbe, 2016-06-07:
On ubuntu machine I removed protoc and did a fresh install using sudo apt-get install protobuf-compiler:

which protoc
/usr/bin/protoc

I get the following error when trying to build.  Can setup.py be changed to look in /usr/local/bin ?

Could not find the protobuf compiler at /usr/local/bin/protoc
On many Linux systems, this is fixed by installing the "protobuf-compiler" and "libprotobuf-dev" packages.
error: [Errno 2] No such file or directory

fahhem, 2016-06-07:
I'm confused, why do you want it to look in `/usr/local/bin/` when `which protoc` is telling you it's in `/usr/bin/`?

wallacbe, 2016-06-07:
the log says setup.py is looking for "usr/local/bin/protoc" which doesn't exist.  Why doesn't setup.py look for "/usr/bin/protoc" which does exist  (was installed with protobuf-compiler)

fahhem, 2016-06-07:
Is this with my change? On ubuntu or mac? On Ubuntu, setup.py only looks in `/usr/local/bin/` if `pkg-config` told it to. Can you paste the output of `pkg-config --variable prefix protobuf` here?

wallacbe, 2016-06-07:
This is from head (with your change) on Ubuntu.

pkg-config --variable prefix protobuf
/usr/local

fahhem, 2016-06-07:
So pkg-config says it's in `/usr/local/bin/`, which is odd. I'll have to look into why `pkg-config` can output the wrong info

wallacbe, 2016-06-07:
Installing from source puts it in /usr/local/bin/protoc, and creates .pc file "/usr/local/lib/pkgconfig/protobuf.pc" which contains prefix=/usr/local, where pkg-config gets its info.

Installing with "sudo apt-get install protobuf-compiler" puts it in /usr/bin/protoc and does not create a .pc file for pkg-config.

Can we use "whereis protoc" to find protoc?

fahhem, 2016-06-07:
Ah, so is there a protobuf.pc file left over from when you installed from source? Because we default to `/usr/bin/` in setup.py for linux, so it should have looked there instead of in `/usr/local/bin/` unless `pkg-config` returned actual data.

We need the prefix, not just the location, because we need to know where the `*.proto` files that came with it are.

wallacbe, 2016-06-07:
Ah ok.  So assuming a fresh machine it will default to /usr/bin.  If it's built from source then it will use  /usr/local/bin.  This works for me.  

To close the loop on this issue I think we should add "protobuf-compiler" installation to CONTRIBUTING.md

Thanks

closed 2016-06-07 by wallacbe