google/openhtf #314: Review and revise package structure.

jettisonjoe, 2016-06-10:
Over time it has become clear that out package structure could use a little more work. Here's a proposal for a reorg. Let's discuss. I'm hoping to get the reorg done before PyPI launch, as it will affect test scripts' imports.

```
.-----------------.
|     openhtf     |
|-----------------|
| Python package. |
'-----------------'
        |
        |    .--------------------------------------------------------------.
        |    |                             core                             |
        |--->|--------------------------------------------------------------|
        |    | Framework internals that manage test state and execution.    |
        |    | The contents of this submodule shouldn't normally need to be |
        |    | accessed from outside the library.                           |
        |    '--------------------------------------------------------------'
        |
        |    .------------------------------------------------------------.
        |    |                           output                           |
        |--->|------------------------------------------------------------|
        |    | Manages the framework's output, including UI, logging, and |
        |    | test records.                                              |
        |    '------------------------------------------------------------'
        |
        |    .-----------------------------------------------------------.
        |    |                           plugs                           |
        |--->|-----------------------------------------------------------|
        |    | Extensions for interfacing with various hardware, such as |
        |    | test equipment and DUT's.                                 |
        |    '-----------------------------------------------------------'
        |
        |    .--------------------------------------------------------------.
        |    |                             util                             |
        |    |--------------------------------------------------------------|
        '--->| Generic utility functions and miscellaneous tools. The       |
             | contents of this submodule should be general enough to be    |
             | useable outside of OpenHTF, meaning it should not be         |
             | dependent on other code in the OpenHTF package itself.       |
             '--------------------------------------------------------------'
```

grybmadsci, 2016-06-16:
+1 this would be awesome.  I like that layout, it's basically exactly what I had in mind.  One thing that's not addressed here that is probably worth bringing up is what our convention is for putting things in a module's `__init__.py` versus in a separate sub-module.

We may also want to do an audit of what goes in `openhtf/__init__.py`, and possibly define `__all__` instead of having the separate `names.py` file (why did we do that in the first place?), as these are the most public-facing spots.

jettisonjoe, 2016-06-16:
I like the idea of auditing `__init__.py` and defining `__all__` instead of using `names`. I created `names` to be a separate, more explicit subset of abstractions to make test writing syntax friendlier because I was slightly concerned importing `*` from `openhtf` would be misleading if you were then still required to import other stuff that isn't part of `__all__`. You might be like "I though I imported *, how come I still have to import [random thing from util] separately?" But maybe the level of Python savvy we're already expecting from test authors is enough to make that a bunk concern.

PR #411 (2016-08-23): Revise package structure as discussed in issue #314.
Fixes #314

See the [original staged PR](https://github.com/jettisonjoe/openhtf/pull/2) for previous review history.

---
This change is [[image]](https://reviewable.io/reviews/google/openhtf/411)
closed 2016-08-23 by jettisonjoe