google/openhtf #315: Define and document a better workflow.

jettisonjoe, 2016-06-11:
In order to release, we should have a good, well-documented workflow for accepting contributions and for managing major/minor/micro releases.

I'm currently leaning towards the one-master flow described [here](http://endoflineblog.com/follow-up-to-gitflow-considered-harmful), but that's not at all set in stone, and I'd like to get some input before I document. I agree with the author's core claim that gitflow is over-engineered in a way that results in a lot of confusion and a history that's pretty hard to follow.

@grybmadsci, @fahhem, any thoughts?

grybmadsci, 2016-06-16:
That single-master workflow LGTM, as well as the "semantic" version numbering scheme.

We should also establish some criteria for PR readiness and release-readiness, such as test coverage and pylint expectations.  Once we get our pylint rules stabilized, I think it's fairly reasonable to expect zero pylint warnings (this forces disable comments, which are easy to catch and call out if we'd like something changed).  As for coverage, I think it's not unreasonable to expect something like 80% coverage, particularly on new code.  For updating existing code, we can say non-negative delta in coverage.

Release readiness right now is probably just all tests pass, since we don't really have a QA process set up.

jettisonjoe, 2016-06-16:
+1 to semantic versioning.

I wonder if we want to add some things to PR readiness like:
  * Rebase; must be able to auto-merge w/ the target branch (master for hotfixes, develop for everything else).
  * Squash spurious commits down to a small set of meaningful ones; not sure whether to include addressing review feedback as separate commits in the final merged PR or squash it down into a commit per salient change at most.

Zero pylint sounds really hard. I'll elaborate on this in #188 in the next few days, but I think there are several places where we disable pylint warnings where "fixing" them would strongly feel like needless hoops to jump through. I think we're gonna have to go through it case by case before we make a final decision on the policy.

80% new code coverage, non-negative delta on modifying existing code sounds very reasonable to me.

On release readiness, I have a server lab here in my building I could use to set up some kind of QA testbed. I have a site server, data server, and some minnows and at least one of those old Lenovo boxes (I could probably find more if I go hunting) that could be set up as a "station". But this is probably for a few months from now once all the current stuff on my plate is stable.

fahhem, 2016-06-21:
I think @grybmadsci meant to say zero pylint warnings, and we'd disable them in the code as needed. I agree that pylint is sometimes dumb or enforces style that doesn't make sense when you want to be a library not a consumer of the library.

If we want to squash commits, GitHub has a feature available at merge time that lets you replace merging branches with a squash. You may need to enable it in the settings, and you can set the default in there as well.

I agree about coverage > 80% and no PR with a delta > -0.5% (you can have small commits that add a comment to an untested function that creates a -0.1% delta).

Regarding gitflow, I used it in a company before and it was wildly confusing to everyone. Every engineer took at least 2 weeks until the finally got the basics of gitflow enough to work without asking me questions, but it took much longer until they could do releases confidently. Whatever we do, let's not do gitflow :D 

Regarding releasing, I think we should have it done automatically, likely based on tags: https://docs.travis-ci.com/user/deployment/releases
We can also deploy to PyPI: https://docs.travis-ci.com/user/deployment/pypi/
I'm not sure if we can deploy to both though :/

jettisonjoe, 2016-06-27:
On releasing, it seems like the common thing to do (and it seems good to me) is to set up PyPI deployment through travis-ci as described in your second link, @fahhem, and to trigger it on tags. Then use GitHub's "releases" tab to create new releases, which will create tags, which will trigger PyPI release through Travis. There's always going to be the manual step of creating a tag, and it seems like GitHub's release creation is essentially a GUI for tag creation, so trying to apply any further automation to _that_ part of it doesn't seem like a huge win.

jettisonjoe, 2016-06-27:
On accepting PR's, I'm leaning towards squashing using GitHub's built-in squash-and-merge PR feature.

On branches, let's try the single-master workflow. If it doesn't seem like it adds too much work to the release process, we could add a convenience branch called 'prod' that always gets fast-forwarded to the most recent tagged release.

grybmadsci, 2016-06-27:
I'm very slightly hesitant on squash-on-merge because it makes partial rollback of large PRs hard.  It's a very slight resistance, though, realistically I think there's much more to be gained than lost by squashing.  I think the "best" solution for the end result would be to "code review" commits and do partial squashes of redundant commits, but that's way more work than any of us really wants to do, so let's roll with squash-on-merge until it actually causes any problems.

@fahhem agree slight negative coverage is okay for very small PRs.  There's some discretion on our party anyway.  Also, yes, I meant zero pylint warnings including disable comments; disable comments stand out so we can be sure that all pylint exceptions are intentional.

Otherwise, everything SGTM, tag-based releases are ideal, I think, and leveraging travis-ci to trigger PyPI launches sounds great.

jettisonjoe, 2016-06-27:
Awesome, sounds like we're all pretty much on the same page.

@grybmadsci, I see where you're coming from with the squash hesitation, and I agree with your _ideal_ solution (and also that it's too much work). I suspect (and hope) that in practice we won't miss that partial rollback ability, especially if we're doing a good job with testing and code reviews.

PR #378 (2016-07-21): Update CONTRIBUTING.md with more workflow details.
Fixes #315
closed 2016-07-21 by jettisonjoe