google/openhtf #316: Support different time formats in output callback

wallacbe, 2016-06-11:
I'd like to add support for different time formats for output call back.  For example we only have start_time_millis available.  How can I display a different time format, like 2016-6-10-10_17:17:23

wallacbe, 2016-06-11:
Is it possible to call time.strftime within the callback?

fahhem, 2016-06-12:
It seems like we could allow the `filename_pattern` to be a callback that takes the `test_record` and returns a string. However, if we do that, I think we should factor it out of `mfg_inspector` and `json_factory` into a helper function that takes an 'old-style' and 'new-style' format string (% and {}), a file-like object, and a callable and returns a string.

grybmadsci, 2016-06-12:
Subclass OutputToJSON to implement your own callback, perform the strftime
in __call__(), open the file yourself, stash the file in
self.filename_pattern, then call super().__call__().  Make sure you restore
output_callback at the end so repeated calls to the callback work right.

wallacbe, 2016-06-12:
Could we add a date/time string to test_record to have it accessible there?  This would allow others to access this if needed instead of having an additional callback.

grybmadsci, 2016-06-13:
the problem with this is that there are a million different string formatting operations that one might want to do besides just a straightforward .format() with the test record for kwargs, and once we start supporting some, it only adds complexity and opens the door for more.

subclassing OutputToJSON is the intended solution if you want JSON output but some feature that's not supported in there, like different formatting for the filename.  If you want to pull out the actual encoding to JSON to a helper function outside `__call__` to make it so you can reuse that code in your own `__call__`, then that would be okay, but i don't really want to start spot-welding random features onto the output callbacks.

grybmadsci, 2016-06-13:
as to @fahhem 's suggestion, that might get some traction if you wanted to establish a base output callback class that both json_factory and mfg_inspector inherited from and make a generic 'output to filename' mechanism, then it's at least in one spot and reused across output modules.

i'd still like to keep it as simple as possible, though.  i could live with supporting a format string for use with .format() or a callback that takes a test record and returns a string, i'd like to deprecate the old %-style strings before PyPI launch if we're going to.

jettisonjoe, 2016-06-14:
@grybmadsci deprecate %-style everywhere? We have a lot of it. Maybe we roll that into our pylint audit.

@fahhem when you say:

> ...a helper function that takes an 'old-style' and 'new-style' format string (% and {}), a file-like object, and a callable and returns a string...

I'm not following with the file-like object. Wouldn't it still make sense to take a test record?

fahhem, 2016-06-14:
No, I meant the helper function (or base class as per @grybmadsci's suggestion) would accept those as options for what is currently named `filename_pattern`. Something like this:

```py
def OpenOutputFilename(pattern_fn_cb, test_record):
  if isinstance(pattern_fn_cb, basestring):
    if '{' in pattern_fn_cb:
      filename = pattern_fn_cb.format(**as_dict)
    else:
      filename = pattern_fn_cb % as_dict
    return open(filename, 'w')
  if callable(pattern_fn_cb):
    return pattern_fn_cb(test_record)
  return pattern_fn_cb
```

In retrospect, a base class with that as a function would be best, as it would allow someone to override that function. We could leave out the `callable` version as well since that could be replaced by subclassing, though the convenience of a lambda cannot be overstated :+1:

jettisonjoe, 2016-06-14:
Ah, I see. Thanks for clarifying.

jettisonjoe, 2016-06-21:
Closing based on in-meeting discussion.

closed 2016-06-21 by jettisonjoe