google/openhtf #317: Flags discussion

wallacbe, 2016-06-13:
Per meeting entering issue for flags discussion.  Establish best practice for using openhtf with other modules/environment that use flags.

One issue is that --help may not display for openhtf if another module is using flags.

jettisonjoe, 2016-06-13:
Taking a step back, though, I think there are two potential problems to think about.

(1) Collisions in the flag namespace. You think you're using an OpenHTF flag, but it _also_ does something you weren't expecting in another library you're using, or vice versa.

(2) If another library is the entry point (third party scheduling, maybe) and it has a `--help` flag that stops execution and they're not using argparse, our flags won't appear in the help, even if they were defined at import.

jettisonjoe, 2016-06-14:
@wallacbe, it would be great if we could get some examples of other libraries that are trying to eat our flags when you get a chance.

grybmadsci, 2016-06-14:
I think (1) is something I'd like to see specific examples of, if there really are collisions, we can rename things accordingly (i'd like to get that renaming done prior to PyPI launch).  I've tried to name flags somewhat OpenHTF-specifically, usually referring to some internal nomenclature rather than simple things like "port", but some might have slipped past.

I also prefer using `conf` values over flags where possible, particularly since they can be overridden via flags if absolutely needed, but also avoid this exact namespace collision problem.  Sometimes flags really do make more sense, but that should only be in very limited cases.

As for (2), there's not much we can do about that, as a library we only have so much control over execution-level stuff like that.  I think good examples and established best-practices are the best we can do.

closed 2016-10-10 by jettisonjoe