google/openhtf #318: Output file syncronization

wallacbe, 2016-06-13:
I made an uploader daemon that monitors files being output to a directory.  There may be a conflict if the daemon tries to access the output file before openhtf output module is done writing.  Should we add a locking mechanism or an atomic write to the output module or should this need to be handled by the daemon?

fahhem, 2016-06-13:
There's code in plumbum for atomic file operations:
http://plumbum.readthedocs.io/en/latest/api/fs.html

I don't think this has to be part of openhtf, but we can refactor the
output methods to allow a subclass to override the writing. I can see the
atomic writer going under `contrib` though.

grybmadsci, 2016-06-14:
Hm, using the plumbum code isn't too terrible of an idea.. a common solution to this problem is actually to write the output to a temp file first, and then os.rename it to the final location when writing is complete (because mv operations are atomic at the filesystem level).  there are a few edge cases where this isn't as cross platform (indeed, in python 3.3, os.replace was introduced to address these), but as long as the destination file doesn't exist, i believe it should work fine.

i think the plumbum code is neat, but rather overkill for this case, where we don't have contention over a shared file (where flock and friends are needed), but rather we want to make sure we write all our output to a single-use file before the next process in the pipeline slurps it up, and a simple os.rename functions for that purpose.

amyxchen, 2016-06-14:
I have a function "atomic_write" in the util, since we have multiple place to use it, it basically write the file to a tmp file, and rename it after it complete writing to be available to others. Not sure if it will fit this case.

grybmadsci, 2016-06-14:
That sounds like exactly what we need, could you send out a PR with just that function @amyxchen and @wallacbe can use it in his new output base class?

closed 2016-08-25 by jettisonjoe