google/openhtf #328: Tag Log Entries in Phases and Plugs

wallacbe, 2016-06-16:
I'd like to add a tag to the log.  I'd like to specify the phase it's from and also have a tag for plugs.  What's the best way to approach this?

Also timestamp doesn't show up in CLI logs.  Any way to enable or control this?

wallacbe, 2016-06-21:
Turning this into generic logger issue.  Also for the case where OpenHTF is calling functions that have their own loggers.  Is there a way to guarantee logging information will show up in the OpenHTF logger?

fahhem, 2016-07-14:
On line 176 and 191 of util.logs, a formatter that includes the time is created and used in the CLI logger. If you want the timestamp instead of the human-readable time, then you'd have to change the formatter on that logger.

One way would be to do this before running any openhtf tests:

```py
logs.SetupLogger()  # It has CallOnce() so it won't override any changes we make here.
logger = logging.getLogger(logs.LOGGING_PREFIX)
for handler in logger.handlers:
  if isinstance(handler, logging.StreamHandler) and handler.stream is sys.stderr:
    handler.formatter = logging.Formatter('your format string')
```

grybmadsci, 2016-07-14:
For now, I suggest manually adding your tag to your log messages.  I was considering a mechanism for correlating log lines to specific phases, but it could theoretically be done post-hoc by looking at respective timestamps.  We can certainly look into auto-prepending phase/plug names to log lines generated.

As for grabbing logs from other places, you might be able to grab the handler from the Test's logger and add it to the logger you want included.  There's no way to guarantee without you explicitly adding the handler to the logger, though, because the library can choose not to propagate its logs to the root logger.  We could make it a config option to include all logs from the root logger in the TestRecord output?  that's going to get pretty chatty, but as long as it's off-by-default i'd be okay with having the option.