google/openhtf #336: Add Images to Prompts

wallacbe, 2016-06-22:
Add ability to display images in prompts.  There are many times when a visual aid is required for an operator.  Sometimes they need to be "active".  An "active" example is showing an operator which keys to press on a DUT and highlight once they are pressed.

grybmadsci, 2016-06-22:
this is a great feature request, but sadly an absolute fuckton of work.  as in, enough to punt it til after PyPI launch, sadly :(  i'd like to overhaul prompts entirely and come up with a more generic solution, perhaps allowing for arbitrary HTML form entry as well, but it's a can of worms.

in the meantime, this week we're hoping to switch prompts from the janky crowbar'd solution we have now to be a plug, which will make it much easier to expand on it later :)  and we have @jettisonjoe 's wonder intern to help with the frontend plugins component that would pair with that.

jettisonjoe, 2016-07-15:
I'm re-framing this as a generic prompt feature overhaul and setting is as a feature planned for the v1.1 release, which is currently planned for end of August.

wallacbe, 2016-09-29:
I have a need where I would like to have a 'PASS' and 'FAIL' operator prompt.

wallacbe, 2016-09-30:
Thoughts on creating a new prompts method in user_input where we use py-getch to poll for a configurable "pass" or "fail" key press then return a boolean result?

def prompts_pass_fail(message, pass_key, fail_key)

fahhem, 2016-09-30:
The prompts are still broken at the moment, so once we fix prompts we can work on more flexible prompts and/or creating prompt-like plugins

fahhem, 2017-02-25:
Prompts are working again, but this will have to be implemented by a custom frontend. As mentioned elsewhere, we're creating one internally that we can contribute back when it's ready, but until then the builtin frontend won't have this feature.

closed 2017-02-25 by fahhem