google/openhtf #337: Configure logging handlers so all loggers show up in openHTF logger and test record

wallacbe, 2016-06-22:
Configure logging handlers so all loggers show up in openHTF logger and test record

fahhem, 2017-02-25:
I do this by first creating the `htf.Test()` object, getting its uid, getting the logger for that uid, then proxying logs into that logger like so:

```py
class PassAlongHandler(logging.Handler):

  def __init__(self, handler):
    self._handler = handler
    super(PassAlongHandler, self).__init__()

  def handle(self, record):
    self._handler.handle(record)

def ProxyLogger(src, dest):
  src.handlers = [PassAlongHandler(dest)]

def my_setup_logs(test_uid):
  testrun_logger = logs.get_record_logger_for(test_uid)
  testrun_base_logger = logging.getLogger(logs.RECORD_LOGGER)
  # Replace openhtf's root handler with a pass-along to the test record.
  openhtf_logger = logging.getLogger('openhtf')
  ProxyLogger(openhtf_logger, testrun_logger)
```

I do that for the 'openhtf' logger, as well as my org's base logger. This does assume only 1 test running in the process at a time, but that's okay for us for now.

silver-kuusik, 2021-02-04:
Hey, when I do htf.Test().uid, I get None. Am I missing something?