google/openhtf #351: Frontend shows "Pass" and doesn't recover from blank DUT ID

wallacbe, 2016-07-03:
If I enter a blank DUT ID the front end will show pass (should be error) and doesn't prompt for another DUT ID
  
File "hello_world.py", line 88, in <module>
    test.Execute(test_start=triggers.PromptForTestStart())
  File "/home/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/__init__.py", line 188, in Execute
    record = self._executor.GetState().GetFinishedRecord()
  File "/home/source/venv/local/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/exe/test_state.py", line 157, in GetFinishedRecord
    'Blank or missing DUT ID, HTF requires a non-blank ID.')
openhtf.exe.test_state.BlankDutIdError: Blank or missing DUT ID, HTF requires a non-blank ID.

jettisonjoe, 2016-10-10:
@wallacbe, Is this still an issue? I think you were going to work on trying to repro this last we talked.

fahhem, 2017-02-25:
Not a problem in master anymore

closed 2017-02-25 by fahhem