google/openhtf #352: Improve DUT result indication in Front End

wallacbe, 2016-07-03:
I setup hello_world to loop.  I ran from head and observed that station history isn't enabled yet (I'm assuming this will go on the side of the page.)   When I enter a DUT ID I see the status go to PASS/GREEN, then go to WAITING/GRAY.  I'm unsure if this is adequate for the operator to know a device's result.  Will they need to look at the history once this is available?  Can we keep the last run DUT's  result at the top of the page until we scan the next DUT?

jettisonjoe, 2016-07-06:
@grybmadsci and I are working on default history in the frontend along with the station API changes, as history depends on the new station API.

fahhem, 2016-11-25:
I believe if you scroll down (or click on History in the future) you'll see previous results now

closed 2016-11-25 by fahhem