google/openhtf #353: Launching frontend discussion and 500: Internal Server Error issue

wallacbe, 2016-07-03:
I have to launch the front end from the root directory, otherwise I get" 500: Internal Server Error"

IOError: [Errno 2] No such file or directory: '/home/source/venv/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/frontend/prebuilt/index.html'

Also is there a way we can import and launch the frontend instead of os.system('python -m openhtf.io.frontend') ?

jettisonjoe, 2016-07-06:
Can you elaborate on root directory, and also list `/home/source/venv/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/frontend/prebuilt/` if it exists?

Yes, I'll clean up the frontend module to enable importing and launching the frontend from Python.

wallacbe, 2016-07-06:
I cloned the repo in ~/source/openhtf  If I launch in ~/source/openhtf it works, if I launch in ~/source I get 

IOError: [Errno 2] No such file or directory: '/home/source/venv/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/frontend/prebuilt/index.html'
ERROR:tornado.access:500 GET / (127.0.0.1) 4.30ms
^CERROR:__main__:Received SIGINT. Stopping frontend server.

~/source$ ls /home/source/venv/lib/python2.7/site-packages/openhtf-0.9-py2.7.egg/openhtf/io/frontend/prebuilt/
fonts  scripts  styles  templates

jettisonjoe, 2016-07-06:
Hmm. Can you list your exact steps to clone and run? I can't repro the error.

wallacbe, 2016-07-19:
cd ~/source/
sudo rm -R openhtf/
git clone https://github.com/google/openhtf.git
source venv/bin/activate
cd openhtf/
python setup.py build
python setup.py install
python -m openhtf.io.frontend <- pass
ctrl + c
cd ..
python -m openhtf.io.frontend <-fail

closed 2017-03-25 by wallacbe