google/openhtf #356: Add timeout to plug methods

wallacbe, 2016-07-11:
I have a case where I would like to execute a method in a plug and have it return a value or raise a timeout exception.

Can we add a timeout decorator to plug method, or add something similar to timeouts.py?

fahhem, 2016-07-14:
Can you use `timeouts.TakeAtMostNSeconds(time_s, func, *args, **kwargs)`? It won't raise an exception, it'll just return `False` though.

grybmadsci, 2016-07-14:
This is tough to do cleanly because of the semantics of the way threads
work (not just in Python, actually).  Threads are fundamentally not
interruptible.  This means you really need to apply your timeout internally
in your plug method.  It's likely you're waiting on some IO operation, like
a read() or write() call, and your timeout needs to be applied there.  If
you're doing a "loop until some condition is true", then you can use the
timeouts.py library as farz suggests (and we could potentially add support
to the LoopUntil* to allow them to be used as decorators).

~madsci

grybmadsci, 2016-07-24:
I suppose as an option, you could start the requested method in a separate thread and then wait on a result in the calling thread, with a timeout.  It's still much better to build the timeout into the plug method itself, because you won't leave hanging threads around, but we can do this, sure.

glados-verma, 2022-10-13:
Not planned; this seems to be a request that can be implemented outside of openhtf and used in your phases/plug implementations.

closed 2022-10-13 by glados-verma