google/openhtf #357: OSX El Capitan Install issues

kdsudac, 2016-07-12:
Few install issues I noticed when installing on OSX El Capitan.

1.  `python setup.py install` command gave an error on openssl headers.  Was able to workaround by:  
    * installing openssl using brew
    * running `pip install cryptography`
2. Building protobuf in `setup.py` fails because check of platform.system() == 'Mac' returns false because platoform.system() returns 'Darwin'.
3.  Additional dependency on protobuf compiler (at least for examples).  Fixed with:
    * `brew install protobuf`
4. To run the all_the_things.py example had to run `python setup.py build` (comment in example suggested running `python setup.py build_proto` which generated the proto py files but didn't copy them to necessary lib folder.

I'll create a PR to fix or improve documentation.

kdsudac, 2016-07-12:
Ha, naturally once I went to create my PR I noticed these instructions which address most of the issues I encountered: https://github.com/google/openhtf/blob/master/CONTRIBUTING.md#macos

grybmadsci, 2016-07-12:
Thanks for the PR - we do try to semi-regularly check to make sure things still work on OSX, but every now and then something slips under the radar :)

The proto dependency in `examples/all_the_things.py` should be only required if you use the `mfg_inspector.py` output callback mechanism, removing that import and commenting out the corresponding line below should get past the proto dependency.

Ultimately, `all_the_things.py` isn't really a friendly example, it's more like our smokescreen integration test we use internally as a sanity check.  We're working to flesh out other examples of each concept/usage separately :)

fahhem, 2016-07-14:
Does this close out with #358?

closed 2016-07-15 by kdsudac