google/openhtf #382: Finish plumbing prompts through to the web gui using station_api.

jettisonjoe, 2016-07-21:


fahhem, 2016-09-23:
Aw dang, I did a bunch of work to migrate to 1.0, then remembered this is broken in 1.0... I'll have to fix this before I can send that out and it changes every single file in my team's purview...

jettisonjoe, 2016-09-23:
I already started a decent portion of the work for this. I can finish it, or you can take it over. I'm on vacation next week, so I likely wouldn't submit my solution until sometime the week after. I'm cool either way. Up to you.

fahhem, 2016-09-23:
Can you send a pr with your initial work so I don't start from scratch?
I'll take over while you're on vacation

jettisonjoe, 2016-09-23:
I sent #443. Turns out it's a small amount of code, even though mentally it felt like a lot of deliberation to arrive at that approach. Open to feedback and alternatives as well.

jettisonjoe, 2016-10-28:
@fahhem I think you guys should have all you need in order to make this work from the Angular side. Hit me up if I'm wrong about that or if you need a hand getting it done.

fahhem, 2016-11-07:
@jettisonjoe Thanks. We were going to start this last week, but something last-minute came up. Hopefully in the next week that'll settle and then we can fix this from the Angular2 side and finally have prompts in 1.0 working :)

fahhem, 2017-02-25:
@Kenadia fixed this a while back

closed 2017-02-25 by fahhem