google/openhtf #384: plug initialization is occurring before the start prompt

wallacbe, 2016-07-24:
plug init is occurring after  serial number prompt.  Below is "all the things".  We should do this prior to ensure the correct equipment state before testing a device.

Provide a DUT ID in order to start the test.
test
init plug

wallacbe, 2016-07-24:
An example is  ensuring that the power supplies are off before inserting a device

grybmadsci, 2016-07-24:
This is actually done intentionally because it may be an arbitrarily long time between calling `.Execute()` and the test start trigger returning, and we wanted to have plugs initialized for only the duration of the test (ie, start time is AFTER the test start trigger returns).  If you need to initialize plugs before a prompt, you can not use test_start and make a phase that prompts for the DUT ID instead.

if we were to change it to initialize plugs before the test start trigger returns, there would be no way to get the current behavior without doing weird things, whereas this way you can manually do the DUT ID phase thing and it's pretty straightforward.

wallacbe, 2016-07-25:
To summarize, we need to pass in a blank dut ID to Execute() and then assign a value to test_record.dut_id in a phase?

When I call Execute() without any arguments I get the following error:
openhtf.exe.test_state.BlankDutIdError: Blank or missing DUT ID, HTF requires a non-blank ID.

Do you recommend I start the tester with the lambda fake serial -> test.Execute(test_start=lambda: 'test')  or should we remove the blank check?  Is starting the test with a fake serial the recommended implementation?

wallacbe, 2016-07-27:
Will this implementation affect the front end?  Looks like it might.

grybmadsci, 2016-07-27:
You should be able to call `.Execute()` with no args, if you can't it's a bug.  That check is supposed to happen only at the end of the test, to give you the chance to set the DUT ID mid-test.

wallacbe, 2016-07-27:
OK it worked as expected.

Only remaining question is if this flow will affect the front end.

Thanks

wallacbe, 2016-07-27:
will this affect the test time calculation and get super inflated?

grybmadsci, 2016-07-27:
yep, right now there's no way to override that.  some options:
 - you could manually update `test_record.start_time_millis`, but i advise against it.
 - you could manually instantiate and tear down your plug to do your initialization before even calling `.Execute()`, but don't forget to `.TearDown()`.
 - each phase has its own `start_time_millis`, so you could still easily calculate the "real test time" as it were, but you'd have artificially inflated test times in mfg-inspector.
 - you could stash a start time in `test.state` and then at the end of your test calculate the test time and store it in a measurement, that's at least more visible than having to dig for the phase `start_time_millis`.

if more use cases come up for it, we can reconsider having the ability to inject plug instances into the test and using that instance instead of instantiating it, but it's a layer of complexity i would prefer to avoid if possible.  that would allow you to do this kind of init pre-`Execute()` more cleanly.

wallacbe, 2016-07-27:
Option #2 sounds the cleanest

grybmadsci, 2016-07-27:
yeah, i agree.  it would maybe be nice if you could then pass that plug in and re-use it instead of re-initializing it, but then the plug is hanging in an "initialized but not being used" state for a long time, so maybe that's bad too.  go with #2 for now and let us know how it works out :)

wallacbe, 2016-07-27:
I currently have my code structured into testers, phases, and plug files.  phases import plugs and testers import phases. This will require me to have testers import phases too.  Just wanted to make a note of that.

grybmadsci, 2016-07-27:
assuming you mean "testers import plugs too", yeah, you're doing some meta-test plug manipulation though, so that makes sense.  "put the fixture in a safe state for DUT entry" isn't really part of "test this DUT", imo, so i think it's actually a reasonable mechanism.  This is why plugs are designed to be easily-instantiated outside the framework, it's expected that you may need to on occasion.

closed 2016-09-15 by jettisonjoe