google/openhtf #397: Add operator/user login

wallacbe, 2016-08-09:
Add operator/user login.  Would add traceability to who is running the test and could enable/disable features or configure permissions based on user privileges.

grybmadsci, 2016-08-09:
hm, at the moment i'm leaning toward the argument that this is more of an OS-level thing, and we could pull the user info from the OS somehow (I believe windows supports the USER environment variable?).

fahhem, 2016-08-09:
In our future environment, the username will always be 'tester', so that's
not useful for knowing who the operator is. I can see this being a plug
though, whose behavior is up to the team, since there are many ways to go
about getting it (require it before the first run, ask for it in parallel
to the first run, use a cli argument, etc)

grybmadsci, 2016-08-10:
yeah, in order to have a true GUI experience, you'd want an actual web-based login that persists across test executions, with the notion of a "shift".  it's getting close to MES-type tracking, though, and it's a bit outside the scope of what we're currently targeting.  I think we're better off focusing on a few things and doing them well right now.

fahhem, 2017-02-25:
This should be implemented using a custom frontend that gathers this information (and stored in a cookie) and sends it to a plug that puts it into the test record (directly or using a phase at the end of the test, or even a teardown phase).

closed 2017-02-25 by fahhem