google/openhtf #403: Plug logger raises AttributeError when called in constructor

wallacbe, 2016-08-17:
plug logger raises AttributeError when called in constructor

added logging statement to example_plug constructor:

 @conf.InjectPositionalArgs
  def __init__(self, example_plug_increment):
    self.logger.info('init')
    self.increment = example_plug_increment
    self.value = 0

AttributeError: 'ExamplePlug' object has no attribute 'logger'
Exception in thread TestExecutorThread:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home/source/venv/local/lib/python2.7/site-packages/openhtf-0.9.2-py2.7.egg/openhtf/util/threads.py", line 59, in run
    self._ThreadProc()
  File "/home/source/venv/local/lib/python2.7/site-packages/openhtf-0.9.2-py2.7.egg/openhtf/exe/__init__.py", line 124, in _ThreadProc
    self.test_state.plug_manager.InitializePlugs()
  File "/home/source/venv/local/lib/python2.7/site-packages/openhtf-0.9.2-py2.7.egg/openhtf/plugs/__init__.py", line 373, in InitializePlugs
    plug_instance = plug_type()
  File "/home/source/venv/local/lib/python2.7/site-packages/openhtf-0.9.2-py2.7.egg/openhtf/conf.py", line 562, in SelfWrapper
    return method_wrapper(**kwargs)
  File "/home/source/venv/local/lib/python2.7/site-packages/openhtf-0.9.2-py2.7.egg/openhtf/conf.py", line 553, in method_wrapper
    return method(**final_kwargs)
  File "/home/openhtf/examples/example_plug.py", line 70, in __init__
    self.logger.info('init')
AttributeError: 'ExamplePlug' object has no attribute 'logger

fahhem, 2016-08-17:
You have to call the super() constructor first, then the logger is
available.

wallacbe, 2016-08-18:
I added super and still observe AttributeError: 'ExamplePlug' object has no attribute 'logger

@conf.InjectPositionalArgs
  def __init__(self, example_plug_increment):
    super(self.__class__, self).__init__()
    self.logger.info('init')
    self.increment = example_plug_increment
    self.value = 0

fahhem, 2016-08-18:
Oh, I didn't check the code. Looks like madsci dropped it in this commit:
https://github.com/google/openhtf/commit/031b14907f87f6c09d91db9889a3ec9d13ca1874
Looks like you added it the first time. Want to add it back?

grybmadsci, 2016-08-18:
The logger isn't accessible in `__init__` because the `PlugManager` sets it.  I considered having the `PlugManager` pass it into `__init__`, but that forces all plug implementations to accept it as an arg, whether they use it in `__init__` or not.

I'd be okay with doing something like a try-except block where you  try to pass a `logger=` argument to the plug's `__init__` and if it raises you simply don't pass it in.  This would allow you to take a `logger` arg in `__init__` and access it directly there.

fahhem, 2016-08-18:
I'd rather avoid the try-except because you might get that error due to some other code and swallow-retry and really confuse folks. Why not use `inspect` to check the args like we do with phases?

grybmadsci, 2016-08-18:
try-except is the Python-recommended way of doing something like this, surprisingly ([EAFP](https://docs.python.org/2/glossary.html#term-eafp)).  catching the right exception (TypeError, I believe?) should be specific enough.  this is, for example, how `hasattr` is implemented.

fahhem, 2016-08-18:
Yeah, and `hasattr()` is confusing when you have an `@property` that fails with
AttributeError but it exists. Such as this:
```py
class A(object): @property def fails_on_hasattr(self): return self.x
hasattr(A(), 'fails_on_hasattr') == False ```
Except you look at the code and sure enough, A has `fails_on_hasattr` but
`hasattr()` says it doesn't! I'd expect `hasattr()` to be True, but `getattr()`
to fail, but that's just me and not the CPython writers, apparently.

On Thu, Aug 18, 2016 7:32 AM, grybmadsci notifications@github.com wrote:
try-except is the Python-recommended way of doing something like this,
surprisingly ( EAFP ). catching the right exception (TypeError, I believe?) should be specific
enough. this is, for example, how hasattr is implemented.

—
You are receiving this because you commented.
Reply to this email directly, view it on GitHub , or mute the thread .

wallacbe, 2016-08-21:
What's the preferred solution?  Do we have to pass a different logger object (other than openhtf.plugs) to __init__ to get the log statement to appear?  The logging statement is useful since it shows when plug __init__ occurs w.r.t. starting a test.

 def __init__(self, plug_types=None, logger=None):
    self._plug_types = plug_types or set()
    self._logger = logger

grybmadsci, 2016-08-21:
You need to use the logger that's held in the PlugManager if you want the log lines to show up in the 'TestRecord'; output.  For debugging/understanding openhtf flow, you can use any old logger.

PR #466 (2016-11-30): Make plug __init__ have a test record logger
Fixes #403
commit a7e11675 (2016-11-30): Make plug __init__ have a test record logger (#466)

Fixes #403
closed 2016-11-30 by fahhem