google/openhtf #404: Mfg inspector output module is adding unset values to failureCodesList

wallacbe, 2016-08-17:
Mfg inspector output module is adding unset values to failureCodesList with details "None".  This is giving incorrect failure codes inspector.  Older openhtf commit I was running would just skip these.

"failureCodesList": [
    {
      "code": "NoDeviceError",
      "details": "No device found"
    },
    {
      "code": "foo",
      "details": "None"
    },
    {
      "code": "bar",
      "details": "None"
    },

wallacbe, 2016-08-17:
Looks like a value variable is getting initialized to None.  It fails the numeric check then is assumed to be a string value and converted to string "None".

wallacbe, 2016-08-17:
Introduced in this commit:

https://github.com/google/openhtf/commit/06a323af55e529d9600b28347afd943e94f097ef

jettisonjoe, 2016-08-17:
Will look shortly.

jettisonjoe, 2016-08-22:
I see two separate issues here, one is described by the issue summary, and the other is described by the subsequent updates.

1. Unset measurements are adding 'None' failure codes. What failure code should they have instead? (PR #405 goes with "Unset").
2. string_value is being set even on unset measurements. PR #405 just won't set this attr at all if the measurement is unset.

wallacbe, 2016-09-20:
This is working now, closing

closed 2016-09-20 by wallacbe