google/openhtf #409: Add Tear Down Function result to test run

wallacbe, 2016-08-22:
Currently tear down function is run outside of the test run record.

I have a case where I'd like to log history and pass/fail of the tear down function.  Can we make this appear in the test run record?

wallacbe, 2016-09-01:
running into another scenario where I am keying off of test_record outcome within the tear down function to run a block of code on pass or fail.  This worked in 0.9, not working in 1.02

jettisonjoe, 2016-09-20:
@fahhem, you were going to look into your flow control concept and propose something that would solve this, right?

grybmadsci, 2016-09-20:
your two comments are inconsistent in the behavior you'd like, I think... the first comment sounds like you want to run the teardown and include its output in the test record, the second comment sounds like you want to look at the result of the test and fork based on that in teardown (which requires the test to be finalized and no longer appendable).

in the latter case, could you do what you're trying to do in an output module instead?  that's really the correct place to make decisions based on test outcome.

wallacbe, 2016-09-20:
1)  Would like the result of the "tear down phase" show up in the test record
2)  Would like to make a decision in the "tear down phase" to put the DUT into a mode based on the test result up until that point.

glados-verma, 2022-10-13:
Not planned - not the intended use case of tear down.

closed 2022-10-13 by glados-verma