google/openhtf #415: Consider replacing our validators with voluptuous

fahhem, 2016-08-24:
https://github.com/alecthomas/voluptuous

Despite the name, it's a data validation library that does everything we need, and more. Its range even lets you specify if the ends should be included as valid or not (fixes another issue if I remember correctly).

As of February, many of the validators are classes just like ours, so our use-cases are 100% covered. What do you think? Worth dropping util/data.py for it?

Mapping:
Range -> Range
MatchesRegex -> Match
Equals -> Match or Range

wallacbe, 2016-09-20:
How much of this would be done under the hood vs author having to know how to use voluptuous and Schema?

fahhem, 2016-09-20:
I was thinking that the validators that we have would get replaced, but the API is the same. So you'd have Range() (or we could rename it to InRange) and Equal() still, but you could have many more as well. Not only would `Range()` be a better one (it allows you to specify whether min/max are included), but we'd have `IsTrue()` instead of `Equals(True)`. It would also open the door for conf keys being validated again, a feature I sorely miss :)

DanLipsitt, 2017-07-26:
If you're still considering this, here are some other validation libraries you may want to consider as well:
https://awesome-python.com/#data-validation

fahhem, 2017-07-27:
I've actually been looking at `py-good` recently, they're a fork of voluptuous with many improvements, including skirting what looks like a contentious naming issue. I haven't had the time to attempt using it yet though

DanLipsitt, 2017-08-22:
It could also be good to have methods that match or use those in [unittest.TestCase](https://docs.python.org/3/library/unittest.html#unittest.TestCase) (aside from the fact that they don't follow pep8 😮 ).