google/openhtf #459: Measurement validators based on arguments

fahhem, 2016-11-14:
The final piece of `WithArgs` (now `with_args`, `format_strings`, etc) would be to make the validators take arguments, too. This has been difficult to nail down because they require type information that the others haven't so far (either taking anything or explicitly stringifying). 

My proposal is to change the name back to `with_args` for consistency and to add it to validators, optionally. If a validators has it implemented, then it will be used. It is then assumed the validator itself will be the right judge for types and if it doesn't have enough context then it should accept the expected type. Validators without it implemented will continue as-is, oblivious to arguments. 

For example:

```py
InRange('{expected_memory.min}', '{expected_memory.max}', type=int) # InRange might take floats or ints, so it needs context
Matches(r'\s+{model_name}\d+') # Matches knows it's a regex so it doesn't bother with types
```

What do you think? Notably, it doesn't make sense to call it `format_strings` in measurements anymore, and it's proliferating my questionable semi-builder-esque naming of `with_args`.

fahhem, 2016-11-14:
I'd assign Keith too, but he's not on here for some reason (or I forgot his username?)

jettisonjoe, 2016-11-15:
My first thought is, weren't we going to change validators to use some third-party library? Or were you imagining doing that a layer below this?

fahhem, 2016-11-15:
I never got a response on the voluptuous stuff. Though I've been following it and it's getting a lot of bug and feature request issues, so I don't know if we want to wait for a more stable release or not. Either way, we could implement this another way and use `singledispatch` to dispatch to either a `with_args()` method or a registered `with_args` converter:

```py
@singledispatch
def validator_with_args(validator, **kwargs):
  ...

@validator_with_args.register(validators.InRange)
@validator_with_args.register(voluptuous.Range)
def range_with_args(inrange, **kwargs):
  ...
  return voluptuous.Range(min=new_min, max=new_max)
```

jettisonjoe, 2016-11-15:
This could be good. Personally I'm in favor of moving toward the third-party validators.

glados-verma, 2022-10-13:
Not planned at the moment. `with_args` is a bit too powerful as it is and reduces readability and discovery.

closed 2022-10-13 by glados-verma