google/openhtf #462: SIGINT at scan prompt results in openhtf.core.test_state.BlankDutIdError and hang

wallacbe, 2016-11-21:
Doing CTRL+C results in BlankDUTId Error and tester hangs (need to press enter to exit)  See issue running all the things

Enter a DUT ID in order to start the test.
^C2016-11-21 10:39:11,531 - ERROR - Received SIGINT, stopping all tests.
2016-11-21 10:39:11,531 - ERROR - Stopping <openhtf.Test object at 0x7f95ed5e5990> due to SIGINT
2016-11-21 10:39:11,531 - ERROR - Test state: <TestState: None@Running Phase: trigger_phase>
KeyboardInterrupt caught, aborting test.
Finishing test with outcome ABORTED.
Traceback (most recent call last):
  File "all_the_things.py", line 124, in <module>
    test.execute(test_start=user_input.prompt_for_test_start())
  File "/home//source/venv/local/lib/python2.7/site-packages/openhtf-1.0.3-py2.7.egg/openhtf/__init__.py", line 240, in execute
    final_state = self._executor.finalize()
  File "/home//source/venv/local/lib/python2.7/site-packages/openhtf-1.0.3-py2.7.egg/openhtf/core/__init__.py", line 95, in finalize
    self.test_state.finalize(test_record.Outcome.ABORTED)
  File "/home//source/venv/local/lib/python2.7/site-packages/openhtf-1.0.3-py2.7.egg/openhtf/core/test_state.py", line 270, in finalize
    'Blank or missing DUT ID, HTF requires a non-blank ID.')
openhtf.core.test_state.BlankDutIdError: Blank or missing DUT ID, HTF requires a non-blank ID.

wallacbe, 2016-11-21:
Also related, I see KeyboardInterrupt when doing CTRL+C when DUT ID has been scanned (which is expected) but also see the hang (have to press enter to exit)

jettisonjoe, 2016-11-23:
I think that should only happen if there's a prompt active at the time you hit CTRL+c. Can you confirm?

wallacbe, 2016-11-23:
CTRL+C at scan prompt: BlankDutIdError and hang
CTRL +C at prompt with blank dut id: BlankDutIdError and hang
CTRL +C at prompt with populated dut id: KeyboardInterrupt and hang
CTRL+C when not in prompt and populated dut id: KeyboardInterrupt and NO hang

fahhem, 2017-02-25:
Should be fixed by #501

closed 2017-02-25 by fahhem