google/openhtf #463: Add a "network_silent" or "local_only" flag.

jettisonjoe, 2016-11-21:
Right now it's a little tricky to disable network activity completely, since you need to set both the `enable_station_discovery` command-line flag to `False`, as well as set the `station_api_port` config value to `None` (really `null` if you're doing it via a config .yaml file). It would be a lot more friendly to provide a single flag to turn off all network activity from OpenHTF.

fahhem, 2016-11-21:
It would also be good to make the test and frontend communicate over
localhost only by default, so you can either enable network discovery or
disable all network activity from there

glados-verma, 2022-10-13:
Closing as stale, this has not been requested recently.

closed 2022-10-13 by glados-verma