google/openhtf #468: Pending/upcoming phases are missing from the frontend

Kenadia, 2016-11-30:
Example:

[image]

We expect to see a pending phase “hello_world” underneath “trigger_phase”.

fahhem, 2016-11-30:
I looked into this, it seems to be that the `TestRecord` only has completed phases in it. I'm going to see if we can pre-populate it with all the phases and replace the relevant one or if we have to do something else (like a separate list of pending phases, which was 0.9's approach)

fahhem, 2017-02-25:
The `StationApi` has been updated to have the test descriptor's phases available. Is that sufficient?

closed 2022-10-13 by glados-verma