google/openhtf #470: Update documentation and examples/measurement.py

csaint00, 2016-11-30:
Ran into some issues on a fresh install from the documentation. I needed to add the package libprotobuf-dev to the apt-get install line in CONTRIBUTING.md to get protobufs to build and got an error when trying to run the example measurements.py that units could not be found, resolved by importing openhtf.utils.units

csaint00, 2016-11-30:
Also ran into a problem cloning the git repo, had to use git clone https://github.com/google/openhtf

jettisonjoe, 2016-11-30:
Oops! Thanks for filing this!

jettisonjoe, 2016-11-30:
Can you elaborate on the cloning issue, though?

csaint00, 2016-11-30:
using "git clone git@github.com:google/openhtf.git" errored out with:

git clone git@github.com:google/openhtf.git
Cloning into 'openhtf'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

jettisonjoe, 2016-11-30:
I _think_ that's an issue with _your_ ssh keys rather than an issue with our repository specifically. Are you able to clone other repo's using ssh? It may just be a matter of generating an ssh key and adding it to your GitHub profile if you haven't yet:

https://help.github.com/articles/adding-a-new-ssh-key-to-your-github-account/

csaint00, 2016-11-30:
Fair enough, I'm more of a hardware guy than software guy so not surprised I'm the problem haha

fahhem, 2016-12-01:
Made the requested changes in #473, let us know how your testing goes!

PR #473 (2016-12-06): Fix bug in examples and CONTRIBUTING.md
Fixes #470
closed 2016-12-06 by fahhem