google/openhtf #474: Launching frontend raises Fault: <Fault 1: "<type 'exceptions.AttributeError'>:'module' object has no attribute 'LoopUntilTimeoutOrNotNone'">

wallacbe, 2016-12-01:
(venv):~/source/openhtf$ python -m openhtf.output.web_gui
Starting openhtf web gui server on http://localhost:12000.
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "openhtf/output/web_gui/__init__.py", line 113, in run
    self._do_callback_for_state(self._test.state)
  File "openhtf/core/station_api.py", line 124, in _wrapper
    return func(*args, **kwargs)
  File "openhtf/core/station_api.py", line 305, in state
    remote_state_dict and RemoteState(cached_state, **remote_state_dict))
  File "openhtf/core/station_api.py", line 185, in __new__
    test_record = pickle.loads(test_record.data)
AttributeError: 'module' object has no attribute '_UnReduceRecord'

Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "openhtf/output/web_gui/__init__.py", line 117, in run
    self._test.wait_for_update(timeout_s=self._wait_timeout_s))
  File "openhtf/core/station_api.py", line 286, in wait_for_update
    self.test_uid, summary_dict, timeout_s)
  File "openhtf/util/xmlrpcutil.py", line 89, in _wrapper
    return method(*args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1306, in single_request
    return self.parse_response(response)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1482, in parse_response
    return u.close()
  File "/usr/lib/python2.7/xmlrpclib.py", line 794, in close
    raise Fault(**self._stack[0])
Fault: <Fault 1: "<type 'exceptions.AttributeError'>:'module' object has no attribute 'LoopUntilTimeoutOrNotNone'">

fahhem, 2016-12-01:
Are the frontend and test running the same version of OpenHTF? It seems they may not be?

wallacbe, 2016-12-02:
Looks like I had another openhtf tester process running.  I rebooted my machine and saw it launch:
python -m openhtf.output.web_gui
Starting openhtf web gui server on http://localhost:12000.
WARNING:tornado.access:404 GET /favicon.ico (127.0.0.1) 3.15ms
WARNING:tornado.access:404 GET /favicon.ico (127.0.0.1) 0.86ms

I then launched a tester binary (which is in another virtual env and not at the same CL where I synced the frontend to)  then tried launching the frontend as above then saw:

Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "openhtf/output/web_gui/__init__.py", line 113, in run
    self._do_callback_for_state(self._test.state)
  File "openhtf/core/station_api.py", line 124, in _wrapper
    return func(*args, **kwargs)
  File "openhtf/core/station_api.py", line 303, in state
    self.test_uid, summary_dict and summary_dict['test_record'])
  File "openhtf/util/xmlrpcutil.py", line 89, in _wrapper
    return method(*args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1306, in single_request
    return self.parse_response(response)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1482, in parse_response
    return u.close()
  File "/usr/lib/python2.7/xmlrpclib.py", line 794, in close
    raise Fault(**self._stack[0])
Fault: <Fault 1: "<type 'exceptions.TypeError'>:a class that defines __slots__ without defining __getstate__ cannot be pickled">

fahhem, 2016-12-07:
I just debugged this sort of problem on someone else's computer. When you run the frontend via `python -m openhtf.output.web_gui` from the root folder, it seems to use the version of openhtf that git is synced to. However, if you run a script that imports openhtf at the top, it seems to instead use what's installed in your virtualenv, which can be a totally different version.

Also, what do you mean by "CL"? I thought you were in git somewhere, not in Perforce land (CL means ChangeList, a Perforce term)?

wallacbe, 2017-03-25:
I don't observe this from head

closed 2017-03-25 by wallacbe