google/openhtf #476: Error message during pip install

mrbaapt, 2016-12-07:
Is there a workaround for this? I am on windows 7.

Installing collected packages: openhtf
  Running setup.py install for openhtf ... error
    Complete output from command C:\Python27\python.exe -u -c "import setuptools, tokenize;__file__='c:\\users\\norm\\appdata\\local\\temp\\pip-build-isvgnd\\openhtf\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record c:\users\norm\appdata\local\temp\pip-xieykd-record\install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_proto
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "c:\users\norm\appdata\local\temp\pip-build-isvgnd\openhtf\setup.py", line 189, in <module>
        'pytest-cov>=2.2.1',
      File "C:\Python27\lib\distutils\core.py", line 152, in setup
        dist.run_commands()
      File "C:\Python27\lib\distutils\dist.py", line 953, in run_commands
        self.run_command(cmd)
      File "C:\Python27\lib\distutils\dist.py", line 972, in run_command
        cmd_obj.run()
      File "build\bdist.win-amd64\egg\setuptools\command\install.py", line 53, in run
      File "C:\Python27\lib\distutils\command\install.py", line 563, in run
        self.run_command('build')
      File "C:\Python27\lib\distutils\cmd.py", line 326, in run_command
        self.distribution.run_command(command)
      File "C:\Python27\lib\distutils\dist.py", line 972, in run_command
        cmd_obj.run()
      File "C:\Python27\lib\distutils\command\build.py", line 127, in run
        self.run_command(cmd_name)
      File "C:\Python27\lib\distutils\cmd.py", line 326, in run_command
        self.distribution.run_command(command)
      File "C:\Python27\lib\distutils\dist.py", line 970, in run_command
        cmd_obj = self.get_command_obj(command)
      File "C:\Python27\lib\distutils\dist.py", line 846, in get_command_obj
        cmd_obj = self.command_obj[command] = klass(self)
      File "C:\Python27\lib\distutils\cmd.py", line 64, in __init__
        self.initialize_options()
      File "c:\users\norm\appdata\local\temp\pip-build-isvgnd\openhtf\setup.py", line 66, in initialize_options
        'Windows support in-progress. Help us by submitting an issue! '
    NotImplementedError: Windows support in-progress. Help us by submitting an issue! https://github.com/google/openhtf/issues/new

    ----------------------------------------
Command "C:\Python27\python.exe -u -c "import setuptools, tokenize;__file__='c:\\users\\norm\\appdata\\local\\temp\\pip-build-isvgnd\\openhtf\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record c:\users\norm\appdata\local\temp\pip-xieykd-record\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in c:\users\norm\appdata\local\temp\pip-build-isvgnd\openhtf\

jettisonjoe, 2016-12-07:
Ah, yes. Sorry about that! This issue was fixed in #454, but we haven't cut a new release from master yet. We have a few pull requests in review for some important frontend fixes, and once those are in we'll likely go ahead and do a versioned release; I would guess sometime next week.

In the meantime, the workaround is to install OpenHTF from source instead of directly from PyPI. Just clone into the repo, make sure you're on the master branch and have pulled all the latest commits, activate your virtualenv, and run `python setup.py install`.

jettisonjoe, 2016-12-07:
It would be great it you could give that source-based install a shot if you have a chance, and let me know if you're still having issues!

mrbaapt, 2016-12-07:
I tried the source install and get this error.
Installed c:\python27\lib\site-packages\openhtf-1.0.3-py2.7.egg
Processing dependencies for openhtf==1.0.3
error: Installed distribution setuptools 2.0.1 conflicts with requirement setuptools>=11.3

jettisonjoe, 2016-12-07:
Are you doing this in virtualenv or system-wide? Looks like you maybe already have an older version of setuptools installed in the Python environment?

mrbaapt, 2016-12-07:
Ok I created a virtualenv and got passed the previous error, now this is the error I get much later in the install process.

Installed c:\users\norm\appdata\local\temp\easy_install-d106ci\cryptography-1.6\.eggs\pycparser-2.17-py2.7.egg
no previously-included directories found matching 'docs\_build'
warning: no previously-included files matching '*' found under directory 'vectors'
_openssl.c
build\temp.win-amd64-2.7\Release\_openssl.c(434) : fatal error C1083: Cannot open include file: 'openssl/opensslv.h': No such file or directory
error: c:\users\norm\appdata\local\temp\easy_install-d106ci\cryptography-1.6\.eggs\cffi-1.9.1-py2.7-win-amd64.egg\_cffi_backend.pyd: Access is denied

jettisonjoe, 2016-12-07:
D'oh! Yeah, the cryptography package is like a second-level dependency or something, and it sucks at installing itself. Try manually installing it this way into your virtualenv:

`pip install --no-cache-dir --only-binary :all cryptography`

jettisonjoe, 2016-12-07:
And after that re-run the openhtf install.

mrbaapt, 2016-12-07:
That did the trick, all good now! Thank you jettisonjoe !

jettisonjoe, 2016-12-07:
Awesome! Happy to help!

closed 2016-12-07 by jettisonjoe