google/openhtf #479: Have to refresh browser to see prompt, get timeout when entering text

wallacbe, 2016-12-19:
Cloned at commit  6981216f1df2b8d6c094b917e6c6b74d2d210b77
Started webgui and then started all_the_things.py

Observed that I'm able to enter barcode through gui to start the test.  I then have to refresh the web browser to see the "What's the widget type" prompt.  I then try to enter text but I then see a timeout in the web server logs:

python -m openhtf.output.web_gui
Starting openhtf web gui server on http://localhost:12000.
ERROR:tornado.application:Uncaught exception POST /plugs/172.19.60.136/8888/13161:1482189226832:139848655111696/openhtf.plugs.user_input.UserInput (127.0.0.1)
HTTPServerRequest(protocol='http', host='localhost:12000', method='POST', uri='/plugs/172.19.60.136/8888/13161:1482189226832:139848655111696/openhtf.plugs.user_input.UserInput', version='HTTP/1.1', remote_ip='127.0.0.1', headers={'Origin': 'http://localhost:12000', 'Content-Length': '73', 'Accept-Language': 'en-US,en;q=0.8', 'Accept-Encoding': 'gzip, deflate, br', 'Host': 'localhost:12000', 'Accept': '*/*', 'User-Agent': 'Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/55.0.2883.87 Safari/537.36', 'Connection': 'keep-alive', 'Referer': 'http://localhost:12000/station/172.19.60.136/8888', 'Content-Type': 'application/json'})
Traceback (most recent call last):
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/tornado-4.3-py2.7-linux-x86_64.egg/tornado/web.py", line 1443, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "openhtf/plugs/__init__.py", line 228, in post
    result = getattr(self.remote_plug, method)(*args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1303, in single_request
    response = h.getresponse(buffering=True)
  File "/usr/lib/python2.7/httplib.py", line 1089, in getresponse
    response.begin()
  File "/usr/lib/python2.7/httplib.py", line 444, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python2.7/httplib.py", line 400, in _read_status
    line = self.fp.readline(_MAXLINE + 1)
  File "/usr/lib/python2.7/socket.py", line 476, in readline
    data = self._sock.recv(self._rbufsize)
timeout: timed out
ERROR:tornado.access:500 POST /plugs/172.19.60.136/8888/13161:1482189226832:139848655111696/openhtf.plugs.user_input.UserInput (127.0.0.1) 3005.17ms
ERROR:tornado.application:Uncaught exception POST /plugs/172.19.60.136/8888/13161:1482189226832:139848655111696/openhtf.plugs.user_input.UserInput (127.0.0.1)
HTTPServerRequest(protocol='http', host='localhost:12000', method='POST', uri='/plugs/172.19.60.136/8888/13161:1482189226832:139848655111696/openhtf.plugs.user_input.UserInput', version='HTTP/1.1', remote_ip='127.0.0.1', headers={'Origin': 'http://localhost:12000', 'Content-Length': '73', 'Accept-Language': 'en-US,en;q=0.8', 'Accept-Encoding': 'gzip, deflate, br', 'Host': 'localhost:12000', 'Accept': '*/*', 'User-Agent': 'Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/55.0.2883.87 Safari/537.36', 'Connection': 'keep-alive', 'Referer': 'http://localhost:12000/station/172.19.60.136/8888', 'Content-Type': 'application/json'})
Traceback (most recent call last):
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/tornado-4.3-py2.7-linux-x86_64.egg/tornado/web.py", line 1443, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "openhtf/plugs/__init__.py", line 228, in post
    result = getattr(self.remote_plug, method)(*args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1303, in single_request
    response = h.getresponse(buffering=True)
  File "/usr/lib/python2.7/httplib.py", line 1089, in getresponse
    response.begin()
  File "/usr/lib/python2.7/httplib.py", line 444, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python2.7/httplib.py", line 400, in _read_status
    line = self.fp.readline(_MAXLINE + 1)
  File "/usr/lib/python2.7/socket.py", line 476, in readline
    data = self._sock.recv(self._rbufsize)
timeout: timed out
ERROR:tornado.access:500 POST /plugs/172.19.60.136/8888/13161:1482189226832:139848655111696/openhtf.plugs.user_input.UserInput (127.0.0.1) 3005.25ms

fahhem, 2016-12-20:
@csaint00, it seems you're not the only one seeing this issue after all!

Kenadia, 2016-12-20:
This problem seems to be related to the fact that plugs are reinitialized after the test start is executed. The problem goes away if we initialize all plugs at once before the test start. Trying now to work toward an appropriate fix.

jettisonjoe, 2016-12-21:
@Kenadia, I don't think this is because plug initialization is happening twice. I haven't actually checked this out, but I _suspect_ it's more to do with the station API being reinitialized after the start trigger when actual test execution starts, which causes a new UUID to be created for the running test instance. Thus the frontend's RemoteTest instance will no longer match the station's representation.

TL;DR -- look to the station API for the cause of this rather than plug init.

grybmadsci, 2016-12-21:
I haven't looked closely, either, but sounds like joe's onto something -
the station api should be singleton for the lifetime of the binary, so if
it's getting re-initialized, that's not intended, and would likely result
in some wonky behavior.

~madsci

wallacbe, 2016-12-21:
Also I'm running Ubuntu 14 + Chrome Browser

Kenadia, 2017-01-07:
After Farz and I investigated, it seems there are two problems:

1. PlugManager's XML-RPC server is killed and restarted after test start, when `initialize_plugs` is called a second time. The result is that the UserInput RemotePlug, which was connected to the killed server, no longer works after test start. This causes the `ERROR:tornado.access:500 POST` error.
2. The Station API is not notified of changes to plug states. Therefore the frontend is not notified of changes to plug states. `StationAPI` has a `wait_for_plug` method but it's not used anywhere. The result is that whether or not a prompt shows up on the front end is a race condition. If the prompt is at the start of a phase, it will usually show up. Below is a short script that demonstrates the problem using a delay.

I've put up PR #484 to solve problem 1 and created a new issue ##483 to track problem 2.

```python
import openhtf
import time

from openhtf.plugs import user_input

@openhtf.plugs.plug(user_input=user_input.UserInput)
def delayed_prompt_phase(test, user_input):
  test.test_record.dut_id = 'my-dut-id'
  print 'Waiting one second.'
  time.sleep(1)
  result = user_input.prompt('Prompt: enter some input', text_input=True)

if __name__ == '__main__':
  openhtf.Test(delayed_prompt_phase, delayed_prompt_phase).execute()
```

wallacbe, 2017-03-25:
I don't observe this from head

closed 2017-03-25 by wallacbe