google/openhtf #507: IndentationError: unindent does not match any outer indentation level

wallacbe, 2017-02-11:
synced from master, observe:

python all_the_things.py 
Traceback (most recent call last):
  File "all_the_things.py", line 25, in <module>
    import openhtf as htf
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-1.0.3-py2.7.egg/openhtf/__init__.py", line 45, in <module>
    from openhtf.core import station_api
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-1.0.3-py2.7.egg/openhtf/core/station_api.py", line 861
    _LOG.debug('Stopped Station API server.')
                                            ^
IndentationError: unindent does not match any outer indentation level

jettisonjoe, 2017-02-11:
PR #506 will fix this, btw.

fahhem, 2017-02-11:
My bad, I did a merge in github and missed that it indented my code by one...

closed 2017-02-11 by fahhem