google/openhtf #508: Front End does not refresh after test complete

wallacbe, 2017-02-12:
I modified "all the things" to launch the front end and loop the tester.

I start the tester, enter the DUT id in the frontend, then the test completes I don't see the front end refresh.  I then refresh the browser manually and enter DUT id, then I see: ERROR:tornado.application:Uncaught exception POST 

Enter a DUT ID in order to start the test.
ERROR:tornado.application:Uncaught exception POST /plugs/192.168.1.37/8888/22750:1486874117439:140192302178576/openhtf.plugs.user_input.UserInput (127.0.0.1)
HTTPServerRequest(protocol='http', host='localhost:12000', method='POST', uri='/plugs/192.168.1.37/8888/22750:1486874117439:140192302178576/openhtf.plugs.user_input.UserInput', version='HTTP/1.1', remote_ip='127.0.0.1', headers={'Origin': 'http://localhost:12000', 'Content-Length': '71', 'Accept-Language': 'en-US,en;q=0.8', 'Accept-Encoding': 'gzip, deflate, br', 'Host': 'localhost:12000', 'Accept': '*/*', 'User-Agent': 'Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/56.0.2924.87 Safari/537.36', 'Connection': 'keep-alive', 'Referer': 'http://localhost:12000/station/192.168.1.37/8888', 'Cookie': 'JSESSIONID=dummy', 'Content-Type': 'application/json'})
Traceback (most recent call last):
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/tornado-4.3-py2.7-linux-x86_64.egg/tornado/web.py", line 1443, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/home/wallaceb/source/venv/local/lib/python2.7/site-packages/openhtf-1.0.3-py2.7.egg/openhtf/plugs/__init__.py", line 243, in post
    result = getattr(self.remote_plug, method)(*args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1303, in single_request
    response = h.getresponse(buffering=True)
  File "/usr/lib/python2.7/httplib.py", line 1089, in getresponse
    response.begin()
  File "/usr/lib/python2.7/httplib.py", line 444, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python2.7/httplib.py", line 400, in _read_status
    line = self.fp.readline(_MAXLINE + 1)
  File "/usr/lib/python2.7/socket.py", line 476, in readline
    data = self._sock.recv(self._rbufsize)
timeout: timed out
ERROR:tornado.access:500 POST /plugs/192.168.1.37/8888/22750:1486874117439:140192302178576/openhtf.plugs.user_input.UserInput (127.0.0.1) 3005.37ms

fahhem, 2017-02-12:
Is this all with a clean install of OpenHTF from master or from PyPI? We've made a ton of fixes in master in the past few days

wallacbe, 2017-02-12:
Master, I cloned few minutes ago

fahhem, 2017-02-12:
can you post your modifications to the example so I can try it?

wallacbe, 2017-02-12:
Put the following after  "test.configure(teardown_function=teardown)"

  import sys, subprocess, openhtf
  subprocess.Popen([sys.executable, '-m', 'openhtf.output.web_gui'])
  while True:
    test.execute(test_start=user_input.prompt_for_test_start())

fahhem, 2017-02-25:
I've sent quite a few pull requests in the past two weeks that hopefully address most of the frontend problems. The remaining issue is that the current frontend doesn't handle history nor clearing of old runs, so it's a bit of a mess. We've forked our frontend off because we need a ton of specific functionality that we don't think would be relevant to general OpenHTF users, but once it's stable we may contribute back a swiss-cheese version of it that should be usable if not as well styled.

I'm closing this for now since I think this particular problem is addressed by one of those pull requests, but open a new issue if you still have issues with all of my pull requests merged in.

closed 2017-02-25 by fahhem