google/openhtf #509: Observe Fault: <Fault 1: "<type 'exceptions.TypeError' running frontend and testers

wallacbe, 2017-02-12:
I'm running bazel environment, starting frontend, and testers on separate threads.  When I run both frontend and tester I see: Fault: <Fault 1: "<type 'exceptions.TypeError'>:a class that defines __slots__ without defining __getstate__ cannot be pickled">

start frontend thread
sleep 5 seconds
Starting openhtf web gui server on http://localhost:12000.
start tester 1 thread
start tester 2 thread
scan tester 1
scan tester 2
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/tmp/tmp.HdtdDlVueu/virtualenv/lib/python2.7/site-packages/openhtf/output/web_gui/__init__.py", line 247, in run
    self._discover()
  File "/tmp/tmp.HdtdDlVueu/virtualenv/lib/python2.7/site-packages/openhtf/output/web_gui/__init__.py", line 234, in _discover
    self.watch_tests(hostport)
  File "/tmp/tmp.HdtdDlVueu/virtualenv/lib/python2.7/site-packages/openhtf/output/web_gui/__init__.py", line 259, in watch_tests
    hostport, remote_test, self._on_update_callback)
  File "/tmp/tmp.HdtdDlVueu/virtualenv/lib/python2.7/site-packages/openhtf/output/web_gui/__init__.py", line 145, in __init__
    self.handle_state_update()
  File "/tmp/tmp.HdtdDlVueu/virtualenv/lib/python2.7/site-packages/openhtf/util/threads.py", line 138, in synchronized_method
    return func(self, *args, **kwargs)
  File "/tmp/tmp.HdtdDlVueu/virtualenv/lib/python2.7/site-packages/openhtf/output/web_gui/__init__.py", line 159, in handle_state_update
    state = self._test.state
  File "/tmp/tmp.HdtdDlVueu/virtualenv/lib/python2.7/site-packages/openhtf/core/station_api.py", line 124, in _wrapper
    return func(*args, **kwargs)
  File "/tmp/tmp.HdtdDlVueu/virtualenv/lib/python2.7/site-packages/openhtf/core/station_api.py", line 320, in state
    self.test_uid, summary_dict and summary_dict['test_record'])
  File "/tmp/tmp.HdtdDlVueu/virtualenv/lib/python2.7/site-packages/openhtf/util/xmlrpcutil.py", line 89, in _wrapper
    return method(*args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1306, in single_request
    return self.parse_response(response)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1482, in parse_response
    return u.close()
  File "/usr/lib/python2.7/xmlrpclib.py", line 794, in close
    raise Fault(**self._stack[0])
Fault: <Fault 1: "<type 'exceptions.TypeError'>:a class that defines __slots__ without defining __getstate__ cannot be pickled">

wallacbe, 2017-02-15:
Tried again from head (commit 1ed1b6a).  Also made sure each thread has its own unique port with station_api_port config.  Any recommendations where I should debug first?  Thanks

fahhem, 2017-02-25:
One thing to watch out for is that you may have a tester binary from another version of OpenHTF running in your network, which can cause weird issues. Unfortunately, versions are not exchanged in the protocol (yet), so it's very possible for version mismatches to cause confusing tracebacks.

This may also be related to attempting to pickle `TestRecord`, which I've just sent a pull request to remove. Hopefully with that PR merged in, you won't see this anymore.

closed 2017-02-25 by fahhem