google/openhtf #510: How to pass multiple config files to testers launched within the same script?

wallacbe, 2017-02-12:
Below is mutithread example for launching multi-up.  Is there a way to pass a config file to each tester instance?  For example I want each tester instance to have it's own "station_id" and unique plug config values.  Or is there another recommended multi-up approach?  Thanks

```
"""Example tester."""

import threading
import time
import openhtf

def Phase1(test):
  test.logger.info('Running Phase1...')

def Phase2(test):
  test.logger.info('Running Phase2...')

def Tester1():
  test = openhtf.Test(Phase1, test_name='Tester1')
  while True:
    test.execute(test_start=lambda: 'serial1')
    time.sleep(1)

def Tester2():
  test = openhtf.Test(Phase2, test_name='Tester2')
  while True:
    test.execute(test_start=lambda: 'serial2')
    time.sleep(1)

if __name__ == '__main__':
  t1 = threading.Thread(target=Tester1)
  t2 = threading.Thread(target=Tester2)
  t1.start()
  t2.start()
```

wallacbe, 2017-02-15:
Approach that I am trying out is to have a "single up" tester script which starts the tester with a Tester() function, and a "multi-up" script which imports the "single-up" Tester() function.

I have a multi-up specific config.yaml file which has config values in a list for all the stations.  The multi-up script iterates through conf._loaded_values, parses, and passes them via --config-value flag to os.system(sys.executable <imported tester>.__file___ blah...).  For each index in the list it will spawn a tester thread.

fahhem, 2017-02-25:
As discussed over video chat, you've changed from doing this in separate threads to doing it in separate processes. Since `conf` is a module, you cannot make it different per-thread (you'd have to do something with threadlocals), but you can make it different per-process.

You've created a launcher binary that takes an input config and generates a new config file per-process, then pass that in using --config-file (or --config-value, but that's discouraged for production use). That's as reasonable an approach as any that we've come up with so far, so I'm mostly documenting this for future folks.

closed 2017-02-25 by fahhem